xdspacelab/openvslam #394: How to change .msg map to pointcloud file like .pcd .obj?

snowless2000, 2020-07-12:
I've never used .msg before and so I want to know is there a way to change it to .obj  file.
I really need some advice.

thanks

YujiElfahkrany, 2020-07-13:
how about using this python script?
https://github.com/xdspacelab/openvslam/issues/87#issuecomment-515683888

snowless2000, 2020-07-15:
> how about using this python script?
> [#87 (comment)](https://github.com/xdspacelab/openvslam/issues/87#issuecomment-515683888)

Thanks for your advice. 
I've read this #87 before, and it really worked, but it only helps me to get the coordinate of each point and I don't know how to build mesh with them. In the file of mesh, it needs more information, not only the coordinates.
I also read some other issues about the mesh building but no one worked.
So I want to know if there is a way to get more informations from the msg or is there a way to build a mesh only with those coordinates .