xdspacelab/openvslam #395: [RGBD Slam] Resume SLAM after getting 'lost'

josealeixopc, 2020-07-13:
I am performing SLAM using the `run_tum_rgbd_slam` command with a custom dataset.

OpenVSLAM tracks nicely the trajectory and surroundings up until a fast camera turn. Then, in the visualizer, it says 'LOST' and no further mapping is done.

How can I tell OpenVSLAM to restart tracking when it gets lost? This way, I could manually "stitch" the resulting trajectories.

Thanks!

dimaxano, 2020-07-13:
Hi! I'm aware of two ways doing that:
1. Push `Reset` button in Pangolin GUI window. 
2. Set [this](https://github.com/xdspacelab/openvslam/blob/6de5fec058647297ca3458e4719b42947fd4a582/src/openvslam/tracking_module.cc#L224) threshold value to 0 and recompile. After that change, tracking will be resetted after each LOST state

Note: after tracking is LOST, the only way to get position on the current map is relocalization. Elsewhere map will be removed and new created (after using methods described above also)

josealeixopc, 2020-07-15:
> Hi! I'm aware of two ways doing that:
>
> 1. Push `Reset` button in Pangolin GUI window.
> 2. Set [...]

Thanks! I will be trying this sometime soon and I'll follow up with how I worked it out. :)

angussmitchell, 2020-08-25:
Just as an update to this, what @dimaxano suggested does not work as the test is always false if **init_retry_thr** is set to 0. 

`curr_frm_.id_ - initializer_.get_initial_frame_id() < camera_->fps_ * init_retry_thr`

I just removed this test from the conditional. The question then becomes, how does one ensure the frame_trajectory.txt, mag.msg and other outputs get written with unique names to ensure the map data is not lost upon reset. 

Any ideas for this?