xdspacelab/openvslam #397: RICOH THETA calibration

kodama-y, 2020-07-16:
Thank you for the excellent algorithm.

Sorry for the elementary question.
I am trying to run OpenVSLAM using RICOH THETA.
But I don't know how to calibrate THETA.
Is there a detailed calibration procedure somewhere?

ymd-stella, 2020-07-19:
If you're using Equirectangular images, no calibration is required.  
dataset is https://openvslam.readthedocs.io/en/master/simple_tutorial.html#equirectangular-datasets.
You can refer to the configuration file included in the dataset.

kodama-y, 2020-07-19:
Thank you for your reply

The camera.name of the setting file is "RICOH THETA S 960". 
The camera I use is the "RICOH THETA Z1". Is there any problem?

ymd-stella, 2020-07-19:
If the same image is given, it works the same way.
try it!

belal-ibrahim, 2020-07-19:
it worked fine with some modification with yaml file according to camera specs.

kodama-y, 2020-07-19:
In other words, does it mean that internal parameters such as the focal length of the spherical camera(RICOH THETA) are not necessary to execute OpenVSLAM?

belal-ibrahim, 2020-07-19:
Try to lookup for the camera model you are using and see the  parameters i.e. resolution of the images. then update these parameters in yaml file. It will  help generate more points.

kodama-y, 2020-07-19:
I understand!

Do you mean that there are only three camera parameters for executing OpenVSLAM: "Camera.fps", "Camera.cols", and "Camera.rows"?

vzhr, 2020-07-20:
The "Equirectangular images" are the images expanded after you capture using RICOH THETA (of course they had been calibrated). U shoulde now how to project the origin image to the  Equirectangular image. The calibration artical reference: "A Flexible Technique for Accurate Omnidirectional Camera Calibration and Structure from Motion", and toolbox reference: http://www-sop.inria.fr/icare/personnel/Christopher.Mei/ChristopherMeiPhDStudentToolbox.html. May this help U.

vzhr, 2020-07-20:
> I understand!
> 
> Do you mean that there are only three camera parameters for executing OpenVSLAM: "Camera.fps", "Camera.cols", and "Camera.rows"?

The "Camera.cols" and "Camera.rows"(Equirectangular images param)  are used to calculate the bearing of keypoints (π/rows, 2π/cols ) + offset. U should expand the RICOH THETA  images ( here has been calibrated) to the "Equirectangular images". 
To calibration the origin images, reference: "A Flexible Technique for Accurate Omnidirectional Camera Calibration and Structure from Motion" ;
Toolbox reference: http://www-sop.inria.fr/icare/personnel/Christopher.Mei/ChristopherMeiPhDStudentToolbox.html.
May this help.

kodama-y, 2020-07-20:
> The "Camera.cols" and "Camera.rows"(Equirectangular images param) are used to calculate the bearing of keypoints (π/rows, 2π/cols ) + offset. U should expand the RICOH THETA images ( here has been [...]

After all, does RICOH THETA have to calibrate?

Where should the parameters calculated by calibration be reflected?

kodama-y, 2020-07-20:
Looking at config.yaml, I thought that OpenVSLAM can be executed if only the resolution of RICOH THETA is known. Is it different?

vzhr, 2020-07-21:
> After all, does RICOH THETA have to calibrate?
>
> Where should the parameters calculated by calibration be reflected?

Are the RICOH THETA images "Equirectangular" images? The openvslam accepts the "Equirectangular" image.

kodama-y, 2020-07-21:
> Equirectangular

The image of RICOH THETA is "Equirectangular".

After all, do I have to find internal parameters (focus length, etc.) even when using RICOH THETA?

kodama-y, 2020-07-21:
> Are the RICOH THETA images "Equirectangular" images? The openvslam accepts the "Equirectangular" image.

The image of RICOH THETA is "Equirectangular".

After all, do I have to find internal parameters (focus length, etc.) even when using RICOH THETA?

vzhr, 2020-07-21:
> > Are the RICOH THETA images "Equirectangular" images? The openvslam accepts the "Equirectangular" image.
> 
> The image of RICOH THETA is "Equirectangular".
> 
> After all, do I have to find internal parameters (focus length, etc.) even when using RICOH THETA?

If so, no need.

kodama-y, 2020-07-21:
> > > Are the RICOH THETA images "Equirectangular" images? The openvslam accepts the "Equirectangular" image.
> > 
> > 
> > The image of RICOH THETA is "Equirectangular".
> > After all, do I have to find internal parameters (focus length, etc.) even when using RICOH THETA?
> 
> If so, no need. "Equirectangular" 是用来算找到的keypoint对应的空间方位角(bearing)，你可以在对应camera的计算中看到，然后你看看你的图像能不能按照对应代码计算公式来计算。如果能就行了，否则就改成对应的或反过来改其计算公式

I understand. Thank you very much.

I have one question, why is it that "Equirectangular" does not need internal parameters?

vzhr, 2020-07-22:
> I understand. Thank you very much.
>
> I have one question, why is it that "Equirectangular" does not need internal parameters?

Because the "Equirectangular"  already include this. The "Equirectangular" is a abstract image.  It has nothing to do with specific camera model. U can search "Equirectangular Projection" for further understand.

kodama-y, 2020-07-22:
> > > > > Are the RICOH THETA images "Equirectangular" images? The openvslam accepts the "Equirectangular" image.
> > > > 
> > > > 
> > > > The image of RICOH THETA is "Equirectangular".
> > > > After all, do I have to find internal parameters (focus length, etc.) even when using RICOH THETA?
> > > 
> > > 
> > > If so, no need. "Equirectangular" 是用来算找到的keypoint对应的空间方位角(bearing)，你可以在对应camera的计算中看到，然后你看看你的图像能不能按照对应代码计算公式来计算。如果能就行了，否则就改成对应的或反过来改其计算公式
> > 
> > 
> > I understand. Thank you very much.
> > I have one question, why is it that "Equirectangular" does not need internal parameters?
> 
> Because the "Equirectangular" already "include" this. The "Equirectangular" is an abstract image. It has nothing to do with specific camera model. U can search "Equirectangular Projection" for further understand.

I see. Thank you very much.