xdspacelab/openvslam #404: fisheye focal_x_baseline setting

asmaaelkeurti, 2020-07-21:
I run the opencv fisheye calibration code. Got the perimeter as below. But stuck on focal_x_baseline. I read the opencv documentation all over again. Still have no clue. Can anyone help me on this?

Camera.name: "None"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 1002.056579
Camera.fy: 999.0578903
Camera.cx: 904.9469381
Camera.cy: 497.8674064

Camera.k1: -0.03192781
Camera.k2: 0.557438388
Camera.k3: -0.91124765
Camera.k4: 0.645854855

Camera.fps: 30.0  
Camera.cols: 1920
Camera.rows: 1080

Camera.color_order: "Gray"  # please reconfigure according to your camera

Camera.focal_x_baseline: ????????

There is another issue:
in run_camera_slam , i notice when opencv open up uvc camera it doesn't specify the resolution. I added two line to set the width and height. And then recompile the project. Is this the right way to fix it?

alvinsunyixiao, 2020-07-21:
If your setup is monocular, the `focal_x_baseline` is unused. You don't have to specify that. Here's a reference explaining what that baseline is in a stereo setup: https://docs.opencv.org/master/dd/d53/tutorial_py_depthmap.html.

Neel125, 2020-09-11:
which camera are you using ? @asmaaelkeurti