xdspacelab/openvslam #406: Fusion of external sensor data in local BA

nfynt, 2020-07-22:
Hello,

I am trying to **fuse the GPS measurement into local BA process by adding a unary g2o edge** to local_keyframe. I am aiming to resolve the scale ambiguity of the monocular slam by introducing GPS measurement as prior.

At the moment it doesn't seem to affect the bundle adjustment process. I have tried following tests:
1. increase the number of keyframe in LBA. To compensate for the low frequency GPS input
2. add the new GPS measurement edge in global BA. However, this doesn't get called since my test data is small (approx. 100 m in straight line) and doesn't involve any loop closing (atm).

I am guessing either the new measurement are getting pruned by outlier detection, or the low frequency (1 Hz) update is unable to stand the optimization of 100 of landmarks and keyframe!

Any ideas on how I can approach this modification in Bundle adjustment process in openvslam?? 

**keywords**: data fusion; local bundle adjustment; scale aware monocular slam;

Best,

urbste, 2020-07-30:
Hi,

I actually started this integration for GoPro GPS data.
You can have a look in my fork: https://github.com/urbste/openvslam/tree/feat/gps-integration

Be aware that I only tested in on two videos, so this is probably still a bit bugy. However the one GoPro sequence was recorded on a mountainbike ride > 2km down a mountain and it worked fine. 

Couple of thoughts what you would need to do:
- Did you align the GPS coordinate frame to the SLAM coordinate frame first? I think without this bootstrapping phase your BA will not converge, if you set XYZ directly as GPSPose vertices (as a prior).  https://github.com/urbste/openvslam/blob/feat/gps-integration/src/openvslam/module/gps_initializer.cc Also if you are just driving on a straight, you will not be able to initialize the BA correctly, as the variance in one coordinate direction is too low. (It's like you could rotate the entire reconstruction around that line and nothing would change.
- The weighting is very important! Usually with GPS the z coordinate is worse than x and y, so I downweighted the z coordinate by a factor of 10. https://github.com/urbste/openvslam/blob/feat/gps-integration/src/openvslam/optimize/global_gps_bundle_adjuster.cc#L100
- The GoPro GPS has a very high frequency, so I interpolated a GPS coordinate for every frame. I don't know about 1hz thought. Also the GoPro GPS is very low-cost so if you want to run this on Kitti data , probably some values needed to be changed

Cheers

dimaxano, 2020-08-04:
Hi, @urbste!
I tried your version of gps fusion and on my examples visual tracking (all three types) failed as soon as gps successfully initialized. How do think maybe we should update `velocity_` matrix as soon gps initialized, cause after GPS initialization map may drastically change its position and orientation.

closed 2020-08-16 by nfynt

asmaaelkeurti, 2020-09-06:
> Hi,
>
> I actually started this integration for GoPro GPS data.
> You can have a look in my fork: https://github.com/urbste/openvslam/tree/feat/gps-integration
>
> Be aware that I only tested in on two [...]

what is your gopro model ?

is it 360 equirectangular or monocular ?

urbste, 2020-09-12:
I used the radial division model I contributed to openvslam: https://github.com/xdspacelab/openvslam/blob/develop/src/openvslam/camera/radial_division.h