xdspacelab/openvslam #409: No image feed

AlejandroSilvestri, 2020-07-27:
[image]

Running localization from tutorial:

    ./run_video_localization -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_2/video.mp4 -c ./aist_living_lab_2/config.yaml --frame-skip 3 --no-sleep --map-db map.msg

openvslam starts, the interface works (pangolin buttons work), it's not frozen, but nothing appears in frame viewer neither map viewer.

No errors in terminal, all files are accessible by those paths: video.mp4 plays nice with totem with that exact path.

I don't know what should appear in frame viewer and map viewer, may be someone with some experience can point me in the right direction.  If someone experienced something like it, please comment whether fixed or not.

Thank you very much.

ymd-stella, 2020-07-27:
```
decoding 0 keyframes to load
decoding 0 landmarks to load
```
Obviously, the map is empty.

You should not skip the [tutorial](https://openvslam.readthedocs.io/en/master/simple_tutorial.html#) instructions. (`Tracking and Mapping` is needed. Is the generated map correct?)
You should do exactly what it says and then ask the question.

AlejandroSilvestri, 2020-07-27:
@ymd-stella 

Thank you for your quick response.  I'm afraid this is not the case.  Those are the exact same instructions I followed. I can see I choose the wrong example.  Here I repeat it with slam instead of localization:

    ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg

[image]

Have you ever had those black windows?  I thought that even with no map, the video input should appear.

AlejandroSilvestri, 2020-08-04:
@dimaxano 

Thank you very much.

With some prints I could check there's no video input and main while exists after the first loop, and I found the app doesn't close because there's no auto-term switch on.

Now i'm adding --auto-term to every command line because this is the expected behaviour.  Video path is fine, may be I have some silent issue with gstreamer or opencv installation.

uhtikiber, 2021-02-16:
Hello! I am experiencing the same issue. Is there a specific solution to this problem?

uhtikiber, 2021-02-19:
I have experienced the same issue. It turned that my opencv was compiled without mp4 support so I recompiled opencv with ffmpeg and everything worked out.

AlejandroSilvestri, 2021-02-19:
It's not openvslam fault, these missing video problems are related to opencv compilation, specifically to the use of gstreamer or ffmpeg.

closed 2021-02-19 by AlejandroSilvestri