xdspacelab/openvslam #413: Triangulation theory

brendanlb, 2020-08-03:
Hi!

Does somebody please give some details/references about the triangulation methods used in `triangulator.h` ?

As I understand it, the first method `static inline Vec3_t triangulate(const cv::Point2d& pt_1, const cv::Point2d& pt_2, const Mat34_t& P_1, const Mat34_t& P_2);` is the classical one, using SVD and presented in Hartley/Zisserman as the Homogeneous method (DLT). 

The third method `static inline Vec3_t triangulate(const Vec3_t& bearing_1, const Vec3_t& bearing_2, const Mat44_t& cam_pose_1, const Mat44_t& cam_pose_2);` is the same, but considers bearing vectors instead of pixel coordinates as input.

I am not sure about the second method `static inline Vec3_t triangulate(const Vec3_t& bearing_1, const Vec3_t& bearing_2, const Mat33_t& rot_21, const Vec3_t& trans_21);`. As I understand it, it is minimizing the distance between 3D points X1 and X2 along the two bearing vectors, where

X1 = lambda1 * bearing_1
X2 = lambda2 * bearing_2

A direct minimization is performed and the triangulated point is in the middle of the line segment [ X1 X2 ].

I could not find where this method is coming from and the theory behind it (when to use it, limitations, etc.). Can somebody give me some explanations or references ?

Thank you very much.

brendanlb, 2020-08-25:
I did some tests and the triangulation results of all methods is very close. Yet, I still don't know where the second triangulation method is coming from.

JackeyLin-ss, 2020-12-23:
@brendanlb the second triangulation method is the same as in REMODE. Checkout line 35 in the following link 
https://github.com/uzh-rpg/rpg_open_remode/blob/master/src/triangulation.cu

brendanlb, 2021-01-05:
Thank you for your answer @JackeyLin-ss. After some research on the Internet, I was able to find out that the second triangulation method is the one called "midpoint triangulation". Some explanation can be found on this [website](https://imkaywu.github.io/blog/2017/07/triangulation/). As the DLT method, the midpoint triangulation is not projective invariant but is also not affine invariant.

closed 2021-01-05 by brendanlb