xdspacelab/openvslam #422: openvslam on mobile ios?

adelpit, 2020-08-13:
Has anyone had any success running openvslam on ios?

SKioresku, 2020-08-14:
Hi! I successfully launched on mobile.

adelpit, 2020-08-14:
> Hi! I was successfully launched on the mobile. I was added some modifications to check initializing and tracking lost state on the fly.

Thanks for the response @SKioresku.  Was this on iOS or android?  I am particularly interested in iOS.  How much work was it to build openvslam and all the dependencies on mobile?

SKioresku, 2020-08-20:
You can compile library with Xcode on the Mac and than add to your project.

fabisant, 2020-08-31:
Hello, Could you help us with the steps you took to compile openVSLAM on the mobile device. I am new to this, and I am working with an application with indoor navigation using augmented reality. could you give us any recommendation?

mcrlc, 2020-10-03:
@SKioresku I'm also curious about it. What did you do to compile it for iOS with all the dependencies?