xdspacelab/openvslam #427: 360 EWquirectangular Video fails to track..

nickponline, 2020-08-26:
I am trying to processing this short time-lapse video (0.5s per frame) and it fails to track 

https://www.dropbox.com/s/ix62g7rvcq8spoy/Timelapse%20Example.zip?dl=0

```
Camera.name: Insta360 ONE X
Camera.setup: "monocular"
Camera.model: "equirectangular"
Camera.fps: 30
Camera.cols: 5760
Camera.rows: 2880
Camera.color_order: "RGB"
Feature.max_num_keypoints: 2000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles: [[0.0, 1.0, 0.0, 0.1], [0.0, 1.0, 0.75, 1.0]]
```

Are there any setting I can use to help it succeed?

ymd-stella, 2020-08-28:
`Feature.max_num_keypoints` per pixel is low.
Try to increase `Feature.max_num_keypoints` to 4000.

nickponline, 2020-08-28:
I've tried up to 20000 with no success

ymd-stella, 2020-08-28:
I've succeeded by
```
./run_video_slam -v orb_vocab.dbow2 -c config.yaml --no-sleep -m video.mp4
```
movie is here. https://www.dropbox.com/sh/n8v5n5vdmt14m1v/AABTIuMY2CVWrcq92YqAoX1Sa?dl=0

nickponline, 2020-08-28:
Ah I'm running on `master`, let me try `develop`.

nickponline, 2020-08-31:
@ymd-stella I realized I linked the wrong video. This is the one that fails including on develop:
https://www.dropbox.com/s/fasem04h7m31w4b/timelapse-fail.mp4?dl=0

```
Camera.name: Insta360 ONE X
Camera.setup: "monocular"
Camera.model: "equirectangular"
Camera.fps: 30
Camera.cols: 3840
Camera.rows: 2160
Camera.color_order: "RGB"
Feature.max_num_keypoints: 4000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7
Feature.mask_rectangles: [[0.0, 1.0, 0.0, 0.1], [0.0, 1.0, 0.75, 1.0]]
```

ymd-stella, 2020-08-31:
Why is the first video 5760x2880 and the next one 3840x2160?
The ratio is different: 5760x2880 is 1:2 and 3840x2160 is 16:9.

nickponline, 2020-08-31:
@ymd-stella It's from the Garmin VIRB 360 Camera which uses 16:9 aspect
ratio rather than a 2:1.

Nick

ymd-stella, 2020-08-31:
> Are there any setting I can use to help it succeed?

I don't think that exists. You should increase the fps.

closed 2020-09-11 by nickponline