xdspacelab/openvslam #429: Current camera pose going backward

gizmosir, 2020-08-27:
Follow the installation step by step and finally run the example successfully. But the **current camera pose are going backard** instead. And the tracking will lose as soon as it detect a looping. Do you have any idea?

[image]
_current_cam_pose are going backward_

[image]
_current_cam_pose look like at the symmetry place of where it should be_

[image]
_before looping is detected, everything seems fine  except the current_cam_pose_

[image]
_as the looping is detected, map change dramatically and tracking no more_

ymd-stella, 2020-08-28:
I think you have the wrong config or input image.
I think it would be a good idea to go to https://spectrum.chat/openvslam and write the information about the camera and the current config.yaml and ask how you should create the config.

gizmosir, 2020-08-29:
@ymd-stella Thank for you advice. I will definitly get it a try. 

Actually I am running the command bellow which pretty much the same as tutorial. And I also tried the KITTI sequence_00 with the config.yaml in example/kitti folder. It has the same problem.

`./build/run_video_slam -v orb_vocab/orb_vocab.dbow2 -c ~/Dataset/equirectangular/aist_living_lab_1/config.yaml -m ~/Dataset/equirectangular/aist_living_lab_1/video.mp4 --frame-skip 3 --map-db aist_living_lab_1_map.msg`

From my point of view, it looks like a hidden bug instead of missing around with dataset and config. I am tracking down this bug and it seems like cam_pose_cw.inverse() bellow do not get the right result.

```
/src/pangolin_viewer/viewer.cc @@185~189

pangolin::OpenGlMatrix viewer::get_current_cam_pose() { 
    const auto cam_pose_cw = map_publisher_->get_current_cam_pose();
    const pangolin::OpenGlMatrix gl_cam_pose_wc(cam_pose_cw.inverse().eval());
    return gl_cam_pose_wc;
}
```

> I am using eigen 3.3.5 and this lib performs correctly at another computer.

ymd-stella, 2020-08-29:
If it's an installation issue, there's little I can advise you on. All you can do for this problem is either match it to the environment we tested it in, or figure out a way to make it work in your environment as well. I'm looking forward to a follow-up report.

surfii3z, 2020-09-03:
Hi @gizmosir 

IDK if it's related or not but when I built g2o and openVSLAM with -DBUILD_WITH_MARCH_NATIVE=OFF it seems to have the same problem.

When I change it to -DBUILD_WITH_MARCH_NATIVE=ON , then it works fine.