xdspacelab/openvslam #430: remove feature over certain distance

asmaaelkeurti, 2020-08-27:
Is there a way to remove the keypoints/feature/landmarks at very far away distance.

I tested openvslam in an outdoor environment. Sometimes there is a tree line at 300 meters away in the background. Openvslam would catch the keypoints from this tree line and then tracking state gets weird. Obviously, these keypoints doesn't move at all because of the distance.

I tried using orb_feature parameter like mask_rectangle to mask the image. it works and i don't think that s the best way to  solve this problem.

any idea?

ymd-stella, 2020-08-28:
> I tested openvslam in an outdoor environment. Sometimes there is a tree line at 300 meters away in the background. Openvslam would catch the keypoints from this tree line and then tracking state gets weird. Obviously, these keypoints doesn't move at all because of the distance.

Without a video and config, it's impossible to understand the problem and give detailed advice.
In particular, whether or not the camera used is monocular is important information.

If sufficient feature points are obtained, outliers that have a significant impact on localization will be removed, so such a removal technique is not necessary.
I think the essence of the problem lies in other aspects such as shooting conditions and feature point extraction/matching.

asmaaelkeurti, 2020-08-31:
> Without a video and config, it's impossible to understand the problem and give detailed advice.
> In particular, whether or not the camera used is monocular is important information.
>
> If sufficient [...]

video file attached.  I was using monocular camera. Please take a look. It was kind of interesting to use openvslam on an robot

[2020-08-30 16-52-24.zip](https://github.com/xdspacelab/openvslam/files/5149664/2020-08-30.16-52-24.zip)

ymd-stella, 2020-09-04:
There appears to be a problem with config.

asmaaelkeurti, 2020-09-06:
> There appears to be a problem with config.

#==============#
# Camera Model #
#==============#

Camera.name: "None"
Camera.setup: "monocular"
Camera.model: "fisheye"

Camera.fx: 1002.056579
Camera.fy: 999.0578903
Camera.cx: 904.9469381
Camera.cy: 497.8674064

Camera.k1: -0.03192781
Camera.k2: 0.557438388
Camera.k3: -0.91124765
Camera.k4: 0.645854855

Camera.fps: 30.0  
Camera.cols: 1920
Camera.rows: 1080

Camera.color_order: "RGB" 

here is the config file.

My config and camera works fine in indoor environment.

Do you think by using stereo camera would be better than fisheye camera for both indoor and outdoor environment?

ymd-stella, 2020-09-06:
The distortion parameters do not appear to be accurate.
How did you calibrate them?
Have you checked the calibration accuracy?