xdspacelab/openvslam #432: ROS 2 colcon build failed

sebastian-ruiz, 2020-08-28:
I followed the ROS 2 installation guide, however on this step:
```
cd /path/to/openvslam/ros/2
colcon buld --symlink-install
```

I receive the following error:
```
openvslam/ros/2 on develop branch
➜ colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> camera_calibration_parsers
Starting >>> image_geometry
Starting >>> image_tools
Finished <<< camera_calibration_parsers [25.8s]                                                                                                                                                               
Finished <<< image_geometry [25.8s]
--- stderr: image_transport                                                                                                           
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp: In constructor ‘image_transport::Publisher::Publisher(rclcpp::Node*, const string&, image_transport::PubLoaderPtr, rmw_qos_profile_t)’:
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp:111:8: error: ‘set’ is not a member of ‘std’
   std::set<std::string> blacklist;
        ^~~
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp:111:8: note: suggested alternative: ‘get’
   std::set<std::string> blacklist;
        ^~~
        get
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp:111:23: error: expected primary-expression before ‘>’ token
   std::set<std::string> blacklist;
                       ^
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp:111:25: error: ‘blacklist’ was not declared in this scope
   std::set<std::string> blacklist;
                         ^~~~~~~~~
/home/sebastian/bin/openvslam/ros/2/src/image_common/image_transport/src/publisher.cpp:111:25: note: suggested alternative: ‘va_list’
   std::set<std::string> blacklist;
                         ^~~~~~~~~
                         va_list
make[2]: *** [CMakeFiles/image_transport.dir/build.make:76: CMakeFiles/image_transport.dir/src/publisher.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:281: CMakeFiles/image_transport.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< image_transport [27.6s, exited with code 2]
Aborted  <<< image_tools [27.6s]
Aborted  <<< cv_bridge [39.4s]                                   

Summary: 2 packages finished [40.0s]
  1 package failed: image_transport
  2 packages aborted: cv_bridge image_tools
  2 packages had stderr output: cv_bridge image_transport
  5 packages not processed

```
Any ideas why it is failing? Thanks for your help.

mirellameelo, 2020-08-28:
Hi! @sebastian-ruiz Which installation guide did you mean?

But, from the information you gave, I believe you are trying to build all the packages together. I had a version with self-contained packages, and I just added it in a branch "ros2-base". 

As follows, the instruction to building it, but in the README file, there is also instruction to run it (look for the topic "Personal notes for running it using ROS2"). Hope it helps you. 

```bh
    # Clone the repository
    cd $HOME
    git clone -b ros2-base --single-branch https://github.com/mirellameelo/openvslam.git
    
    #Build Vision openCV and Image Common packages
    source /opt/ros/<DISTRO>/setup.bash
    cd $HOME/openvslam/ros2
    colcon build --packages-select cv_bridge camera_calibration_parsers image_geometry image_transport opencv_tests vision_opencv camera_info_manager
    
    # Source Vision openCV and Image Common packages
    source $HOME/openvslam/ros2/install/setup.bash
    
    # Build openvslam project
    cd $HOME/openvslam
    mkdir build && cd build
    cmake     -DBUILD_WITH_MARCH_NATIVE=OFF     -DUSE_PANGOLIN_VIEWER=ON     -DUSE_SOCKET_PUBLISHER=OFF     -DUSE_STACK_TRACE_LOGGER=ON     -DBOW_FRAMEWORK=DBoW2     -DBUILD_TESTS=ON     ..
    make -j4
    sudo make install
    
    # Build openvslam packages
    cd $HOME/openvslam/ros2
    colcon build --symlink-install
```

sebastian-ruiz, 2020-08-31:
I am refering to the installation guide on the develop branch: https://openvslam.readthedocs.io/en/master/ros_package.html#installation

mirellameelo, 2020-08-31:
Ok. So, you are trying to build all packages together, right?  But notice that Openvslam packages depend on those, so I recommend you to build first the dependencies, and then the Openvslam packages.

First this:
> colcon build --packages-select cv_bridge camera_calibration_parsers image_geometry image_transport opencv_tests vision_opencv camera_info_manager

And only after **sourcing** them, build the Openvslam packages (openvslam and publisher).

sebastian-ruiz, 2020-09-02:
@mirellameelo thanks for your help. I looked at your branch. I follow all the commands exactly ([see here](https://github.com/sebastian-ruiz/openvslam-ros2-docker)) from your personal notes and everything goes smoothly until the last command `colcon build --symlink-install` where I get weird errors to do with `publisher`:
```
root@31c8af7d26cd:~/openvslam/build# cd $HOME/openvslam/ros2/
root@31c8af7d26cd:~/openvslam/ros2# colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> camera_calibration_parsers
Starting >>> image_geometry
Finished <<< image_geometry [1.50s]
Finished <<< camera_calibration_parsers [2.40s]
Finished <<< cv_bridge [2.60s]
Starting >>> opencv_tests
Starting >>> vision_opencv
Finished <<< opencv_tests [0.55s]
Finished <<< vision_opencv [0.73s]
Finished <<< image_transport [3.81s]
Starting >>> camera_info_manager
Starting >>> openvslam
Starting >>> publisher
Finished <<< camera_info_manager [0.88s]
--- stderr: publisher
In file included from /opt/ros/dashing/include/rclcpp/subscription.hpp:42:0,
                 from /opt/ros/dashing/include/rclcpp/callback_group.hpp:26,
                 from /opt/ros/dashing/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/dashing/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/dashing/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/dashing/include/rclcpp/executor.hpp:33,
                 from /opt/ros/dashing/include/rclcpp/executors/multi_threaded_executor.hpp:24,
                 from /opt/ros/dashing/include/rclcpp/executors.hpp:21,
                 from /opt/ros/dashing/include/rclcpp/rclcpp.hpp:144,
                 from /root/openvslam/ros2/src/publisher/src/image_publisher.cc:1:
/opt/ros/dashing/include/rclcpp/subscription_traits.hpp:79:19: error: ‘enable_if_t’ in namespace ‘std’ does not name a template type
   typename = std::enable_if_t<!std::is_integral<
                   ^~~~~~~~~~~

... OMIT A LOT OF LINES with similar type of errors to do with rclcpp ...

make[2]: *** [src/CMakeFiles/video_publisher.dir/video_publisher.cc.o] Error 1
make[1]: *** [src/CMakeFiles/video_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [src/CMakeFiles/image_publisher.dir/image_publisher.cc.o] Error 1
make[1]: *** [src/CMakeFiles/image_publisher.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< publisher [2.71s, exited with code 2]
Aborted  <<< openvslam [5.84s]

Summary: 7 packages finished [7.46s]
  1 package failed: publisher
  1 package aborted: openvslam
  2 packages had stderr output: openvslam publisher
root@31c8af7d26cd:~/openvslam/ros2#
```

For reproducibility I put it in a docker container: https://github.com/sebastian-ruiz/openvslam-ros2-docker

mirellameelo, 2020-09-03:
@sebastian-ruiz 

I just updated the [Docker](https://github.com/mirellameelo/openvslam/blob/ros2-base/Dockerfile.desktop) file. It worked for me. Could you please try on your machine?

After built, run with:

```bh
docker run -it --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix:ro openvslam-desktop
```

And then:
```bh
source /opt/ros/dashing/setup.bash
cd $HOME/openvslam/ros2
colcon build --symlink-install
```

mirellameelo, 2020-09-03:
@sebastian-ruiz 

I just updated the [Docker](https://github.com/mirellameelo/openvslam/blob/ros2-base/Dockerfile.desktop) file. It worked for me. Could you please try on your machine?

```bh
# to build
cd path/to/dockerfile
docker build -t openvslam-desktop -f Dockerfile.desktop .
# after built, run
docker run -it --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix:ro openvslam-desktop
```

sebastian-ruiz, 2020-09-03:
Thank you!

Your dockerfile seems to work except for the last line
```
RUN /bin/bash -c "source /ros_entrypoint.sh && cd $HOME/openvslam/ros2 && colcon build --symlink-install"
```
which fails, but this can be run after sh'ing into the container.

When I run the example:
Terminal 1:
```
ros2 run publisher video -m $HOME/openvslam/videos/aist_living_lab_1/video.mp4
```

Terminal 2:
```
ros2 run openvslam run_slam -v $HOME/openvslam/Vocabulary/orb_vocab/orb_vocab.dbow2 -c $HOME/openvslam/videos/aist_living_lab_1/config.yaml
```

I get the following problem in terminal 2:
```
root@c1973afad8e9:/# ros2 run openvslam run_slam -v $HOME/openvslam/Vocabulary/orb_vocab/orb_vocab.dbow2 -c $HOME/openvslam/videos/aist_living_lab_1/config.yaml
[2020-09-03 14:08:17.767] [I] config file loaded: /root/openvslam/videos/aist_living_lab_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2020-09-03 14:08:17.768] [I] loading ORB vocabulary: /root/openvslam/Vocabulary/orb_vocab/orb_vocab.dbow2
[2020-09-03 14:08:18.167] [I] startup SLAM system
[2020-09-03 14:08:18.167] [I] start global optimization module
[2020-09-03 14:08:18.167] [I] start mapping module
error: XDG_RUNTIME_DIR not set in the environment.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X11 Error: BadMatch (invalid parameter attributes)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to create an OpenGL context
```
so I think I have some problem with X11, but Openvslam itself is running!

I am running docker on Ubuntu 20.04.

sebastian-ruiz, 2020-09-07:
I fixed this problem:
```
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X11 Error: BadMatch (invalid parameter attributes)
X11 Error: BadValue (integer parameter out of range for operation)
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Failed to create an OpenGL context
```
by adding `--runtime=nvidia` and having nvidia-docker2 installed.

sebastian-ruiz, 2020-09-07:
When running the example I get a problem:

Terminal 1: 
```
# ros2 run publisher video -m $HOME/openvslam/videos/aist_living_lab_1/video.mp4
2020-09-07 13:32:05.672 [RTPS_WRITER Error] Error sending fragment (459, 83) -> Function send_any_unsent_changes
2020-09-07 13:32:05.672 [RTPS_WRITER Error] Cannot add RTPS submesage to the CDRMessage. Buffer too small -> Function insert_submessage
2020-09-07 13:32:05.672 [RTPS_WRITER Error] Error sending fragment (459, 84) -> Function send_any_unsent_changes
```

Terminal 2:
```
# ros2 run openvslam run_slam -v $HOME/openvslam/Vocabulary/orb_vocab/orb_vocab.dbow2 -c $HOME/openvslam/videos/aist_living_lab_1/config.yaml --debug
[2020-09-07 13:46:29.522] [D] CONSTRUCT: config
[2020-09-07 13:46:29.522] [I] config file loaded: /root/openvslam/videos/aist_living_lab_1/config.yaml
[2020-09-07 13:46:29.522] [D] load camera model type
[2020-09-07 13:46:29.522] [D] load camera model parameters
[2020-09-07 13:46:29.522] [D] CONSTRUCT: camera::base
[2020-09-07 13:46:29.522] [D] CONSTRUCT: camera::equirectangular
[2020-09-07 13:46:29.522] [D] compute image bounds
[2020-09-07 13:46:29.522] [D] load ORB parameters
[2020-09-07 13:46:29.523] [D] load tracking parameters
[2020-09-07 13:46:29.523] [D] load depth threshold
[2020-09-07 13:46:29.523] [D] load depthmap factor
[2020-09-07 13:46:29.523] [D] CONSTRUCT: system
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2020-09-07 13:46:29.523] [I] loading ORB vocabulary: /root/openvslam/Vocabulary/orb_vocab/orb_vocab.dbow2
[2020-09-07 13:46:29.916] [D] CONSTRUCT: data::camera_database
[2020-09-07 13:46:29.916] [D] CONSTRUCT: data::map_database
[2020-09-07 13:46:29.916] [D] CONSTRUCT: data::bow_database
[2020-09-07 13:46:29.917] [D] CONSTRUCT: publish::frame_publisher
[2020-09-07 13:46:29.917] [D] CONSTRUCT: publish::map_publisher
[2020-09-07 13:46:29.917] [D] CONSTRUCT: module::initializer
[2020-09-07 13:46:29.917] [D] CONSTRUCT: module::relocalizer
[2020-09-07 13:46:29.917] [D] CONSTRUCT: tracking_module
[2020-09-07 13:46:29.917] [D] CONSTRUCT: mapping_module
[2020-09-07 13:46:29.917] [D] CONSTRUCT: global_optimization_module
[2020-09-07 13:46:29.917] [I] startup SLAM system
[2020-09-07 13:46:29.917] [I] start mapping module
[2020-09-07 13:46:29.917] [I] start global optimization module
error: XDG_RUNTIME_DIR not set in the environment.
[2020-09-07 13:46:30.166] [D] CONSTRUCT: initialize::bearing_vector
[2020-09-07 13:46:30.259] [I] initialization succeeded with E
[2020-09-07 13:46:30.259] [D] DESTRUCT: initialize::bearing_vector
```
When running the command in terminal 2, the PangolinViewer opens and then immediately closes.

mirellameelo, 2020-09-07:
Great news, @sebastian-ruiz!

There is a note [here](https://openvslam.readthedocs.io/en/master/simple_tutorial.html) that says: "If OpenVSLAM terminates abnormally soon after initialization, rebuild g2o and OpenVSLAM with -DBUILD_WITH_MARCH_NATIVE=OFF option for CMake configuration."

Once I had this problem too, in this [Docker](https://github.com/mirellameelo/openvslam/blob/ros2-base/Dockerfile.desktop) I already used this configuration, so please double-check it in the DockerFile you're using.