xdspacelab/openvslam #434: Map-database (*.msg) extracted using msgpack python doesn't give correct localized path.

geekykant, 2020-08-30:
I am able to generate the localized mapping of the demo area accurately. However as per #87 The .csv plot that was generated lies in small fractional values.

The coordinates were in ranges of X in (-9, 8) and Y in(-8,1). I cannot exactly replicate the same localized map.
What's the get through?

**Plotting on Sheet**
[image]

**SocketViewer**
[image]

fabisant, 2020-08-31:
You need to georeference the coordinates, you can use CloudCompare. You need to have at least 4 reference points.
https://www.youtube.com/watch?v=S32wy66e_0o

JoaoAVDuarte, 2020-11-18:
> The coordinates were in ranges of X in (-9, 8) and Y in(-8,1)

In your 1st image the coordinates are mostly within that range though, the ones that don't fit in that range are some outliers that can be easily removed using something like Open3D: 
```python
import open3d as o3d

# Read data as o3d pointcloud
# ...

pcd, ind = pcd.remove_radius_outlier(nb_points=5, radius=0.1)

# ...
```

ymd-stella, 2020-11-18:
This issue is caused by a lack of understanding of map.msg and the data in it.
This is not an OpenVSLAM issue.
If you want to treat it as an OpenVSLAM issue, you should propose a feature request to document the use of map.msg.
Otherwise, it should be handled at https://spectrum.chat/openvslam.

closed 2020-11-19 by geekykant