xdspacelab/openvslam #435: Merging multiple non-overlapping maps

0rchon, 2020-08-31:
I modified the openVSlam algorithm that I get separate map-files if the tracking is lost for a certain number of frames. The problem that I am facing now is that I want to merge these map-files together to get one big map. In my scenario the maps are also not overlapping, which would make solving the problem a little bit easier. One of the constraints is, that I know the sequence of the maps, because it is operating on video-files.

My question would be, if someone else is having an idea or can point me to a way to solve my problem at hand.

I know that my question is not solely OpenVSlam specific, but maybe some one else is having the same issue as I am and hs already figured something out.

Thanks in advance

ayushGHub, 2020-12-03:
> I modified the openVSlam algorithm that I get separate map-files if the tracking is lost for a certain number of frames. The problem that I am facing now is that I want to merge these map-files together to get one big map. In my scenario the maps are also not overlapping, which would have made solving the problem a little bit easier. One of the constraints is, that I know the sequence of the maps, because it is operating on video-files.
> 
> My question would be, if someone else is having an idea or can point me to a way to solve my problem at hand.
> 
> I know that my question is not solely OpenVSlam specific, but maybe some one else is having the same issue as I am and has already figured something out.
> 
> Thanks in advance

Could you share the changes you have made to generate separate map-files, and how are those maps separated? Is it different maps whenever there is tracking loss or something else?

Thanks

dimaxano, 2020-12-05:
Hi, @0rchon 
I think that map merging is possible only (especially if they are not overlapped) if you have some sort of global information how those maps corresponds to each other
But if there are some overlapping I would advice to look at ORB SLAM 3 techniques on how to map merging is done

0rchon, 2020-12-06:
First, thank you really much for the responses. In the meantime since I posted this question, I moved on from this topic and I am working on other topics now, but maybe the discussion of this problem will help other developers in the future.

@ayushGHub Unfortunately, I can't share the code (would really like to). But the results that I have are like when you would start the openvslam algorithm all over again. With the change that you restart the algorithm on the specific lost frame from the previous run. So the maps are completely seperated from each other.

@dimaxano In the scenario that I was facing, I did not have any overlapping. But I did make some assumptions (completely aware, that I could get completely false results if they were not fulfilled). For example I have two trajectories from openvslam, then my idea would have been:

- continuation of the first trajectory for the next n frames after the lost frames (for example via kalman filter)
- scaling the second trajectory according to first trajectory (with average speed and distance between frames from first trajectory)
- using the (simulated) continuation of the trajectory for overlapping and merging the two trajectories