xdspacelab/openvslam #438: why origin keyframe is not (0,0,0)?

Kushine, 2020-09-03:
In openvslam, origin keyframe is seted (0,0,0), and pose is fixed in BA. but the  keyframe_trajectory.txt saved  show that the first keyframe is not (0,0,0), why?
0 2.71269749 -0.76365717 3.49601326 0.012870079 0.0577311957 -0.0106234621 0.998192673
0.567233351340136 2.7126609 -0.767077622 3.49806058 0.009455766 0.0645174166 -0.010751783 0.997813856
0.600600019066026 2.7125735 -0.768158387 3.49842158 0.00833845315 0.0631492569 -0.010702853 0.997911865
0.800800025421369 2.71198896 -0.776754901 3.49988663 -0.00124219892 0.0477294349 -0.0110220886 0.998798714
0.90090002859904 2.71159 -0.781393998 3.50035727 -0.00700895968 0.0428920371 -0.0117732147 0.998985755
1.0343666995026 2.71165461 -0.787975669 3.5012642 -0.0152262333 0.0439635018 -0.0119777752 0.998845286
1.13446670268027 2.711264 -0.793594646 3.50158646 -0.0211159246 0.0423222212 -0.0118165686 0.998810951
1.56823338311685 2.70732598 -0.818056057 3.5021342 -0.0271313208 0.0694916082 -0.00809500335 0.997180665
3.70370011757383 2.705387 -0.8970064 3.49996942 -0.0281965581 0.0561617354 0.0403946428 0.997205639
4.80480015252821 2.70143427 -0.911468234 3.5097767 -0.0941946156 0.074044151 -0.00924305247 0.992753446
8.94226695053861 2.71054097 -0.939183259 3.53104913 -0.0427337194 0.0381500021 0.0380684399 0.997631796
11.578233700884 2.71303789 -0.937631332 3.54303732 -0.171846625 -0.00431589579 0.0360447676 0.984454613
13.3133004226303 2.71868133 -0.934219863 3.53242382 -0.169261338 -0.163252457 0.106962065 0.966052975
13.7137004353409 2.7036548 -0.939841823 3.5054129 -0.0843978177 -0.0956485528 0.0738426582 0.989078169
13.9806337771481 2.69463449 -0.939805459 3.51200731 -0.0339104497 -0.0940503668 0.0304616258 0.994523353
14.6813337993918 2.69539882 -0.940149621 3.54039416 -0.102993311 0.00532084248 -0.0283512497 0.994263684
17.0837338756559 2.70207106 -0.947665089 3.56439448 -0.0774547099 -0.00790250364 -0.00436500926 0.996954996
18.4517672524174 2.7173783 -0.956522972 3.72912262 -0.0739185762 -0.0176622627 -0.00190392967 0.997106044
18.5852339233209 2.71607197 -0.95539915 3.74965751 -0.07395342 -0.0191668836 0.00027782738 0.997077452
18.8855339328539 2.71689079 -0.955038705 3.80418538 -0.074028355 -0.0170773569 -0.00187876159 0.997108137
20.086733970986 2.71959622 -0.953040713 4.02920316 -0.0693347763 -0.0109947741 -0.0117825803 0.99746327
21.0877340027627 2.73130807 -0.95676858 4.21207634 -0.0751360273 -0.0402706756 -0.0143119915 0.996257003
22.4891340472501 2.73420412 -0.960930206 4.4854706 -0.0961286068 -0.0422313317 -0.0097855199 0.994424481
23.0230007308643 2.73657178 -0.961760986 4.58640952 -0.0782374035 -0.0521413506 -0.0141210612 0.995470132
23.2899340726715 2.73791503 -0.959046576 4.63573037 -0.0719267398 -0.0472880347 -0.014570156 0.996181759
24.2575674367223 2.74077175 -0.960418754 4.82503747 -0.0620921826 -0.0421363471 -0.00845589368 0.997144717
24.5245007785294 2.74213048 -0.963562046 4.87743757 -0.0665287596 -0.0412209728 -0.00691061434 0.996908721
25.7257008166615 2.74637807 -0.961684371 5.12775283 -0.0763993229 -0.0377744235 -0.00614212208 0.996342567
27.2605675320524 2.76514237 -0.966113345 5.44526585 -0.0744428762 -0.0299009276 -0.0139217733 0.996679676
27.694334212489 2.77235906 -0.964141683 5.54298603 -0.0963547637 -0.0367643473 -0.0148217943 0.994557417
27.9612675542961 2.77431043 -0.966707405 5.59579585 -0.110422592 -0.0299373631 -0.0109489152 0.993373407
28.8288009151693 2.77658472 -0.969611336 5.76340457 -0.0980907986 -0.0543172229 -0.0029290132 0.993689718
29.5962342728647 2.77538469 -0.968449093 5.9227182 -0.0877874419 -0.0720740794 0.0025580418 0.993525112
30.0633676210272 2.77137357 -0.966785025 6.0233987 -0.082028948 -0.0613844527 0.00299777753 0.994733238
30.5638676369156 2.76597098 -0.969849403 6.12901017 -0.0841235715 -0.0627459335 -0.000983525531 0.994477353
......

angussmitchell, 2020-09-06:
This looks like expected behavior to me- The first timestamp is not 0, and thus the first coordinate, and rotations would not be zero. If you like, you could always add a first line to the output, forcing the pose to be 0, at time 0:
`0, 0, 0, 0, 0, 0, 0, 0`

Kushine, 2020-09-07:
The first timestamp is 0，but the pose is 2.71269749 -0.76365717 3.49601326 0.012870079 0.0577311957 -0.0106234621 0.998192673. as you see in 
0 2.71269749 -0.76365717 3.49601326 0.012870079 0.0577311957 -0.0106234621 0.998192673
0.567233351340136 2.7126609 -0.767077622 3.49806058 0.009455766 0.0645174166 -0.010751783 0.997813856
0.600600019066026 2.7125735 -0.768158387 3.49842158 0.00833845315 0.0631492569 -0.010702853 0.997911865
0.800800025421369 2.71198896 -0.776754901 3.49988663 -0.00124219892 0.0477294349 -0.0110220886 0.998798714
0.90090002859904 2.71159 -0.781393998 3.50035727 -0.00700895968 0.0428920371 -0.0117732147 0.998985755
1.0343666995026 2.71165461 -0.787975669 3.5012642 -0.0152262333 0.0439635018 -0.0119777752 0.998845286
1.13446670268027 2.711264 -0.793594646 3.50158646 -0.0211159246 0.0423222212 -0.0118165686 0.998810951
1.56823338311685 2.70732598 -0.818056057 3.5021342 -0.0271313208 0.0694916082 -0.00809500335 0.997180665
3.70370011757383 2.705387 -0.8970064 3.49996942 -0.0281965581 0.0561617354 0.0403946428 0.997205639
4.80480015252821 2.70143427 -0.911468234 3.5097767 -0.0941946156 0.074044151 -0.00924305247 0.992753446
8.94226695053861 2.71054097 -0.939183259 3.53104913 -0.0427337194 0.0381500021 0.0380684399 0.997631796
11.578233700884 2.71303789 -0.937631332 3.54303732 -0.171846625 -0.00431589579 0.0360447676 0.984454613
13.3133004226303 2.71868133 -0.934219863 3.53242382 -0.169261338 -0.163252457 0.106962065 0.966052975
13.7137004353409 2.7036548 -0.939841823 3.5054129 -0.0843978177 -0.0956485528 0.0738426582 0.989078169
13.9806337771481 2.69463449 -0.939805459 3.51200731 -0.0339104497 -0.0940503668 0.0304616258 0.994523353
14.6813337993918 2.69539882 -0.940149621 3.54039416 -0.102993311 0.00532084248 -0.0283512497 0.994263684
17.0837338756559 2.70207106 -0.947665089 3.56439448 -0.0774547099 -0.00790250364 -0.00436500926 0.996954996
18.4517672524174 2.7173783 -0.956522972 3.72912262 -0.0739185762 -0.0176622627 -0.00190392967 0.997106044
18.5852339233209 2.71607197 -0.95539915 3.74965751 -0.07395342 -0.0191668836 0.00027782738 0.997077452
18.8855339328539 2.71689079 -0.955038705 3.80418538 -0.074028355 -0.0170773569 -0.00187876159 0.997108137
20.086733970986 2.71959622 -0.953040713 4.02920316 -0.0693347763 -0.0109947741 -0.0117825803 0.99746327
21.0877340027627 2.73130807 -0.95676858 4.21207634 -0.0751360273 -0.0402706756 -0.0143119915 0.996257003
22.4891340472501 2.73420412 -0.960930206 4.4854706 -0.0961286068 -0.0422313317 -0.0097855199 0.994424481
23.0230007308643 2.73657178 -0.961760986 4.58640952 -0.0782374035 -0.0521413506 -0.0141210612 0.995470132
23.2899340726715 2.73791503 -0.959046576 4.63573037 -0.0719267398 -0.0472880347 -0.014570156 0.996181759
24.2575674367223 2.74077175 -0.960418754 4.82503747 -0.0620921826 -0.0421363471 -0.00845589368 0.997144717
24.5245007785294 2.74213048 -0.963562046 4.87743757 -0.0665287596 -0.0412209728 -0.00691061434 0.996908721
25.7257008166615 2.74637807 -0.961684371 5.12775283 -0.0763993229 -0.0377744235 -0.00614212208 0.996342567
27.2605675320524 2.76514237 -0.966113345 5.44526585 -0.0744428762 -0.0299009276 -0.0139217733 0.996679676
27.694334212489 2.77235906 -0.964141683 5.54298603 -0.0963547637 -0.0367643473 -0.0148217943 0.994557417
27.9612675542961 2.77431043 -0.966707405 5.59579585 -0.110422592 -0.0299373631 -0.0109489152 0.993373407
28.8288009151693 2.77658472 -0.969611336 5.76340457 -0.0980907986 -0.0543172229 -0.0029290132 0.993689718
29.5962342728647 2.77538469 -0.968449093 5.9227182 -0.0877874419 -0.0720740794 0.0025580418 0.993525112
30.0633676210272 2.77137357 -0.966785025 6.0233987 -0.082028948 -0.0613844527 0.00299777753 0.994733238
30.5638676369156 2.76597098 -0.969849403 6.12901017 -0.0841235715 -0.0627459335 -0.000983525531 0.994477353
......

Kushine, 2020-09-07:
@angussmitchell

angussmitchell, 2020-09-07:
Appologies @Kushine, I did not see that for some reason. 
I agree that at time = 0, translations and rotations should be 0 too. Hopefully the owners will amend this. 

As a suggested fix for the short term, you could subtract the initial translation and rotation off all values in the file.