xdspacelab/openvslam #439: Innacuract result of mono video SLAM

angussmitchell, 2020-09-04:
I've been encountering strange and plainly incorrect results from running the video_slam example. I ran it on the "aist_living_lab_1" example, and plotting using [evo tool](https://github.com/MichaelGrupp/evo/wiki), the rotation results are clearly incorrect, and do not reflect the values seen through the viewer.

The x y z translations look accurate:
[image]

However the rotation values are very wrong:
[image]

This video only really contains rotations in one dimension (yaw - about the y axis), however in the frames_trajectory.txt, you can see that there are significant rotations about the z and x axes (pitch and roll). 

These plots were created by running 
`evo_traj tum frame_trajectop _0.txt -p`
on the frames_trajectory.txt output.

MichaelGrupp, 2020-09-09:
You probably need to change the Euler angle representation used for this plot. This is not related to accuracy of the trajectory.
https://github.com/MichaelGrupp/evo/issues/295#issuecomment-649584091
https://github.com/MichaelGrupp/evo/issues/181#issuecomment-507675295

angussmitchell, 2020-09-09:
Thanks very much @MichaelGrupp, I see that this is a common issue - that there is a mismatch between the Euler angle representation that openvslam outputs in the frames_trajectory.txt file, and what is expected by evo by default. 

I am working on an application where I want to stabilize video based on rotation about a single axis (with x y or z), which cannot be predicted (i.e. its not always about the y or the z axis in the case of a camera mounted to a car). 

From the previous issues raised it seems that the output is represented in szxy rotation order. That being the case, using standard libraries such as scipy how does one convert the openvslam outputs to axis angles? 

i.e. using scipy I get the same result as the default evo sxyz rotation order

`from scipy.spatial.transform import Rotation as R
r = R.from_quat([1, 0, 0, 0])`

I understand that there is a configurable rotation order in evo, but surely openvslam should output a rotation order than is compatible with the convention expected by users and by other modules such as scipy. It seems that several past issues have raised the same problem. Is it possible to set the default output order to correspond with what users expect?

MichaelGrupp, 2020-09-25:
I don't understand - the trajectory file contains a quaternion, not RPY. So if you want to convert it to an Euler representation, you need to choose the order yourself.