xdspacelab/openvslam #440: yaml configuration for E-con Tara stereo camera

Neel125, 2020-09-08:
I have a E-con Tara stereo camera how can i setup config file for this camera. 
and also when i try to run ./run_camera_slam then it is using 2 cameras but i only plugged in E-con Tara stereo camera. so, how can I differentiate left and right camera of E-con Tara stereo camera.

mirellameelo, 2020-09-10:
Hi. Once ROS has support for Tara, I believe [this tutorial](https://www.e-consystems.com/Articles/Camera/ros-support-for-tara.asp) on how to publish the stereo images together with (this one)[http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration] about stereo camera calibration can help you.

Neel125, 2020-09-11:
Thanks for your reply. 
But i think camera mentioned in tutorial is not supported for Jetson Nano. is this tutorial useful jetson nano ?

mirellameelo, 2020-09-11:
@Neel125 
To calibrate your camera, this is, to get the parameters of it, the brand, the model is not the point. A good camera for this project is simply one that works - between the types supported: monocular, stereo, or RGBD. What are your prerequisites? For what you said, you want to use a stereo camera and also having the project running in the Jetson Nano, am I right? (My answer is based on that).
So, given that, I'd recommend you to:
(1) install ROS in your Jetson Nano - ([here a tutorial](https://www.stereolabs.com/blog/ros-and-nvidia-jetson-nano/). Notice that I'm supposing your Jetson Nano is powered by Ubuntu 18.04. If it is not, then you have to install the proper ROS version;
(2) calibrate your camera. To make it via ROS, two steps are necessary:
           - 2.1: publish the image (in your case, two images). Check out [this another tutorial](https://www.e-consystems.com/blog/camera/interfacing-e-con-cameras-with-ros/) for that. It works for ROS melodic in Ubuntu 18.04 and ROS Kinetic in Ubuntu 16.04;
           - 2.2: run the camera calibration package from ROS (it should work for both ROS kinetic and melodic, so you are covered). [The tutorial](http://wiki.ros.org/camera_calibration/Tutorials/StereoCalibration) is the same as I mentioned previously;
(3) With the config file for your stereo E-con camera, build the Openvslam project ([here the turorial](https://github.com/xdspacelab/openvslam/blob/master/docs/ros_package.rst), but use the project provided [here](https://github.com/ymd-stella/openvslam/tree/feat/ros-stereo). It is a branch that supports a stereo cameras and ROS. 

Hope you succeed in that.

closed 2020-09-17 by Neel125