xdspacelab/openvslam #442: Keyframe selection

mirellameelo, 2020-09-11:
Hi!

As far as I know, visual SLAM defines if a frame will become a keyframe based on the quantities of landmarks detected compared to the last keyframe. This is how OpenvSLAM works in order to add a new keyframe? If so, there is any way in which I could use a different threshold regarding the landmarks quantities to determine a keyframe selection? I'm not sure how feasible it is... Any guidance on how to do it inside the OpenvSLAM project is really helpful.

Thanks in advance.

dimaxano, 2020-09-13:
Hi!
Have a look at this function to get a better understanding of when OpenVSLAM decides to insert a new keyframe
https://github.com/xdspacelab/openvslam/blob/9eb280aa4aad6105450debf2c85fbfd477d811f5/src/openvslam/module/keyframe_inserter.cc#L25

P.S. `develop` branch has comments translation

mirellameelo, 2020-09-15:
Thank you, @dimaxano. This is exactly what I was looking for!!

Unfortunately, I'm really confused about the conditions. Two of them seem to be redundant if I understood it right. I did a graphical representation to help my understanding that I'll share here. I'm wondering if one of the authors could clarify it... 

[image]

[image]

[image]

1) In order to find the quantity of tracked landmarks visualized by the current frame (_num_tracked_lms_) I believe all local landmarks are considered, not only the 3D points saw by the last keyframe. I concluded it by this function which increments the _num_tracked_lms_ variable.

https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/src/openvslam/tracking_module.cc#L343

2) The A1, A2, A3, and B condition (I divided B in B1 and B2 for best understanding) are all necessary to create a new KF, this is because all of them are verified by "&&", so all must be true. Then, why the verification is separated? By the comments, "B is mandatory", but actually, all conditions are mandatory, aren't they?

3) Given that all the condition is mandatory, A3 and B2 are the same equation, but once you satisfy A3 you consequently also satisfy B2, so is this not redundant?

4) The same happens for A1 and A2. Once you satisfy the maximum threshold automatically you satisfy the minimum threshold... Really didn't get this...

This is my interpretation after reading the code and would be great to make sure I'm not making this up...

mirellameelo, 2020-09-15:
Thank you, @dimaxano. This is exactly what I was looking for!!

Unfortunately, I'm really confused about the conditions. Two of them seem to be redundant if I understood it right. I did a graphical representation to help my understanding that I'll share here. 

[image]

[image]

[image]

Consideration: In order to find the quantity of tracked landmarks visualized by the current frame (_num_tracked_lms_) I believe all local landmarks are considered. I concluded it by this function which increments the _num_tracked_lms_ variable.

https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/src/openvslam/tracking_module.cc#L343

1) A3 and B2 have the same equation, but once you satisfy A3 you consequently also satisfy B2, so is this not redundant?

2) The same happens for A1 and A2. Once you satisfy the maximum threshold automatically you satisfy the minimum threshold... Really didn't get this...

This is my interpretation after reading the code and would be great to make sure I'm not making this up... Thanks in advance.

mirellameelo, 2020-09-17:
Well, this is a funny thing... Totally unusual of my logical thinking. rsrsr All those conditions are actually NOT desired. WHich means that the system just add a new KF if condition B and some of the conditions A are FALSE.
At the end, I had to invert all equations to be able to think rsrsrs, and also change the final condition to:

```
    if (cond_b && mapper_is_idle && (cond_a1 || cond_a2 || cond_a3)) 
        return true;
```

closed 2020-09-18 by mirellameelo

ymd-stella, 2020-09-20:
> but I think neither A1 and A3 are necessary once A2 is always true because the minimum value (min_num_frms_) is set to 0

mapper_is_idle must be satisfied in order for A2 to be satisfied.

mirellameelo, 2020-09-21:
@ymd-stella, thanks for replying!
Yeah, you're right! I noticed that too, but _mapper_is_idle_ is double-checked, so I'm not sure what's the reasons behind check twice this var in specific. Besides, even with this extra condition, A2 (for my scene at least) was always true. It was not only true when a new KF was added, but always that a new frame was checked...

ymd-stella, 2020-09-22:
It is not double-checked because abort_local_BA may be called when A1 and A3 are true and mapper_is_idle is false.