xdspacelab/openvslam #444: docker run error   run_video_slam

flydragon2018, 2020-09-15:
root@c7759471b5c4:/openvslam/build# ./run_video_slam -v orb_vocab/orb_vocab.dbow2 -c aist_livint_lab_1/config.yaml -m aist_living_lab_1/video.mp4 --frame-skip 3 --map-db aist_living_lab_1_map.msg
yaml-cpp: error at line 0, column 0: bad file

Except that It took many times to successfully build the docker, every step seems ok until running example.

Ubuntu 18.04

ymd-stella, 2020-09-16:
The file (`aist_livint_lab_1/config.yaml`) does not appear to exist.

flydragon2018, 2020-09-18:
my bad for the typo. however, after I fixed it,  I got the new error:

[2020-09-18 10:54:58.665] [I] config file loaded: aist_living_lab_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: LUMIX DMC-GX8 8mm fisheye 1K
- setup: Monocular
- fps: 30
- cols: 960
- rows: 540
- color: RGB
- model: Fisheye
  - fx: 441.73
  - fy: 442.521
  - cx: 480.357
  - cy: 275.49
  - k1: -0.030687
  - k2: -0.00334345
  - k3: -0.00288174
  - k4: -0.000591742
  - min x: -1688.94
  - max x: 2616.09
  - min y: -968.628
  - max y: 1343.02
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7

[2020-09-18 10:54:58.666] [I] loading ORB vocabulary: orb_vocab/orb_vocab.dbow2
terminate called after throwing an instance of 'std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >'
*** Aborted at 1600426498 (unix time) try "date -d @1600426498" if you are using GNU date ***
PC: @     0x7fa3ae8ecf47 gsignal
*** SIGABRT (@0x14) received by PID 20 (TID 0x7fa3b27acc00) from PID 20; stack trace: ***
    @     0x7fa3b1b188a0 (unknown)
    @     0x7fa3ae8ecf47 gsignal
    @     0x7fa3ae8ee8b1 abort
    @     0x7fa3af16d957 (unknown)
    @     0x7fa3af173ae6 (unknown)
    @     0x7fa3af173b21 std::terminate()
    @     0x7fa3af173d54 __cxa_throw
    @     0x7fa3b1dc7528 DBoW2::TemplatedVocabulary<>::loadFromBinaryFile()
    @     0x7fa3b1dc4bcc openvslam::system::system()
    @     0x555cdcc89c61 (unknown)
    @     0x555cdcc88965 (unknown)
    @     0x7fa3ae8cfb97 __libc_start_main
    @     0x555cdcc88e2a (unknown)
Aborted (core dumped)

jawadSajid, 2020-09-18:
@flydragon2018 Any luck solving this issue?

flydragon2018, 2020-09-18:
not yet. don't know the reason

flydragon2018, 2020-09-18:
Could some one who had run successfully, post manually  installation in detail? 

thanks.

ymd-stella, 2020-09-18:
orb_vocab/orb_vocab.dbow2 does not appear to exist.
Please read [Simple Tutorial](https://openvslam.readthedocs.io/en/master/simple_tutorial.html?highlight=orb_vocab).

ymd-stella, 2020-09-18:
These questions should first be submitted to https://spectrum.chat/openvslam.

flydragon2018, 2020-09-19:
The file is there, I am sure, I have checked.  why It complains about date issue?

ymd-stella, 2020-09-20:
> why It complains about date issue?

This is a message from glog and has nothing to do with this issue.

ymd-stella, 2020-09-20:
Is `/openvslam/build/orb_vocab/orb_vocab.dbow2` definitely present?

jawadSajid, 2020-09-29:
Docker container seems to delete the orb_vocab.dbow2 and the yaml file, this is why I downloaded them on my local system. And used the -v flag to make a data folder in my docker container containing the files, this was done as follows:
 "sudo docker run --rm -it --name openvslam-socket --net=host -v /<path to dir>/openvslam/data:/data/ openvslam-socket" and "./run_video_slam -v /data/orb_vocab/orb_vocab.dbow2 -m /data/mytest.mp4 -c /data/mytest.yaml"
This solved all the problems for me. @ymd-stella @flydragon2018