xdspacelab/openvslam #446: Running video is very slow in Jetson nano

Neel125, 2020-09-16:
Hello, 
i am using jetson nano to run openvslam and uses stereo camera for it. it is worked well but it is running very slow in jetson nano.

i am using config parameters like this,
# intrinsic parameters after stereo-rectification are written

Camera.name: "temp"
Camera.setup: "stereo"
Camera.model: "perspective"

# new "rectified" matrices is the first three cols of the projection matrix which is calculated with cv::stereoRectify()
# e.g. fx = P1[0][0] or P2[0][0], cx = P1[0][2] or P2[0][2]
#      fy = P1[1][1] or P2[1][1], cy = P1[1][2] or P2[1][2]

Camera.fx: 435.2046959714599
Camera.fy: 435.2046959714599
Camera.cx: 367.4517211914062
Camera.cy: 252.2008514404297

# there is no distortion after stereo rectification

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

# focal_x_baseline is -P2[0][3] which is calculated with cv::stereoRectify()

Camera.fps: 30.0
Camera.cols: 640
Camera.rows: 480
Camera.focal_x_baseline: 47.90639384423901

Camera.color_order: "Gray"

#======================#
# Stereo Rectification #
#======================#

# original intrinsic parameters (K, D) and stereo-recitification parameters (R)
# matrices (K, R) are written in row-major order

StereoRectifier.K_left: [458.654, 0.0, 367.215, 0.0, 457.296, 248.375, 0.0, 0.0, 1.0]
StereoRectifier.D_left: [-0.28340811, 0.07395907, 0.00019359, 1.76187114e-05, 0.0]
StereoRectifier.R_left: [0.999966347530033, -0.001422739138722922, 0.008079580483432283, 0.001365741834644127, 0.9999741760894847, 0.007055629199258132, -0.008089410156878961, -0.007044357138835809, 0.9999424675829176]
StereoRectifier.K_right: [457.587, 0.0, 379.999, 0.0, 456.134, 255.238, 0.0, 0.0, 1]
StereoRectifier.D_right: [-0.28368365, 0.07451284, -0.00010473, -3.555907e-05, 0.0]
StereoRectifier.R_right: [0.9999633526194376, -0.003625811871560086, 0.007755443660172947, 0.003680398547259526, 0.9999684752771629, -0.007035845251224894, -0.007729688520722713, 0.007064130529506649, 0.999945173484644]

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 1000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#========================#
# Initializer Parameters #
#========================#

Initializer.num_min_triangulated_pts: 100

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.07
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.65
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400

ymd-stella, 2020-09-16:
There is not enough information in this description.
What is the dataset you used?
What exactly is very slow?

Neel125, 2020-09-17:
Thanks for your replay. @ymd-stella 
I am running openvslam in jetson nano using my custom video and this video was recorded with stereo camera. After running ./run_camera_slam, the mapping video is running very slow like approx 4-5 FPS. And also it takes more time for tracking like 200-300 ms.

ymd-stella, 2020-09-18:
I think it's just a problem of CPU and memory performance.
You will need to use a profiler to investigate.

Neel125, 2020-09-22:
Hi @ymd-stella ,
I did monitoring of RAM, CPU, GPU usage in Jetson nano. I ran on video file and it was using all cpus upto 90%, GPU 30-40% 
and RAM 75%.

surfii3z, 2020-09-22:
> Hi @ymd-stella ,
> I did monitoring of RAM, CPU, GPU usage in Jetson nano. I ran on video file and it was using all cpus upto 90%, GPU 30-40%
> and RAM 75%.

I am wondering why does your openvslam use GPU??

Neel125, 2020-09-22:
@surfii3z 
because i built opencv with cuda.

surfii3z, 2020-09-23:
> @surfii3z
> because i built opencv with cuda.

That's interesting. I have been trying to port openvslam to CUDA too (but not successful so far).

May I ask which part of the pipeline did you change to GPU computation?

Neel125, 2020-09-25:
> > @surfii3z
> > because i built opencv with cuda.
> 
> That's interesting. I have been trying to port openvslam to CUDA too (but not successful so far).
> 
> May I ask which part of the pipeline did you change to GPU computation?

@surfii3z 
Compile opencv with CUDA

surfii3z, 2020-09-25:
> > > @surfii3z
> > > because i built opencv with cuda.
> > 
> > 
> > That's interesting. I have been trying to port openvslam to CUDA too (but not successful so far).
> > May I ask which part of the pipeline did you change to GPU computation?
> 
> @surfii3z
> Compile opencv with CUDA

Yeah, I did exactly that, but it doesn't seem to use GPU directly without modifying the source code. Please kindly suggest me if I misunderstood anything.

flyaudio, 2020-10-13:
Is it the "cv::FAST()" that you have changed to CUDA version ?