xdspacelab/openvslam #448: acess key points value

roboticsai, 2020-09-23:
I am trying to modify the slam and localization code so that i can remove the pangolin and socket viewer and instead use ros pointcloud2 data type to integrate openvslam properly in ros1. But neither i saw the api documentation of openvslam nor any proper method how to do it i.e retrieve the x,y,z value of every key points calculated in the process of slam and localization.
So can anyone tell me what is the class or function which can be used to retrieve the x,y,z value of every key points extracted during slam and localization process?

mirellameelo, 2020-09-23:
I believe the function you're looking for is this one:

https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/src/openvslam/data/landmark.cc#L30

To have access to x,y,x of any landmarks in your ros code, you can add a function like this into system.cc (At least I did this way...):

     void system::global_landmarks(std::vector<openvslam::data::landmark*>& all_landmarks){
          all_landmarks = map_db_->get_all_landmarks();
      }

roboticsai, 2020-09-23:
It looks like this function needs a static map already created by openv slam process. What is the method which can access the landmark in real time i.e while running the slam process itself. Because my goal is to access those landmarks in real time while openvslam slam process runs and publish them as sensor_msgs/PointCloud2 ros data type. And also it looks their is no API documentation for openvslam. How are developers going to use openvslam functions in this way?

mirellameelo, 2020-09-23:
I believe the function you're looking for is this one:

https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/src/openvslam/data/landmark.cc#L30

(Sorry, I delete by mistake). So, this function can be called even during the process. It will return whatever map points are created by that time.

roboticsai, 2020-09-23:
Thanks. I'll try it.

mirellameelo, 2020-09-23:
@roboticsai 
[In this file](https://github.com/mirellameelo/openvslam/blob/8a27a9e61916eaa25817d38aa0c9aaf0eb13bc43/ros2/src/openvslam/src/run_slam.cc#L1), and in this specific commit, you can find an example of publishing odometry and point cloud during slam process. (It is works for ros2, but I believe it help you...)