xdspacelab/openvslam #449: OpenVSLAM on ROS is compiled and can be invoked but not visual, when executing examples.

mayur444607, 2020-09-23:
Hello,

OpenVSLAM on ROS is compiled on Nano and can be invoked but no visuals of openvslam mapping UI, when executing examples.

I successfully install OpenVslam on jetson nano.
Link: https://openvslam.readthedocs.io/en/master/overview.html#installation
OpenVslam is running as expected.
Then I tried to install and run openvslam on ros melodic version, installation is successfully.
Link: https://openvslam.readthedocs.io/en/master/ros_package.html
While running example with publisher and subscriber no visual data on slam is showing.
Publisher is of ROS and Subscriber is of openvslam.

When analyzed code found that ‘openvslam Subscriber’ is handling in run_slam.cc file, from that file the ‘mapping module.cc’ and ‘global optimization module.cc’ file is calling when subscriber is in running process. But further couldn’t continued in code debugging, so can anybody please guide me how to run OpenVSLAM using ROS on Nano?

https://user-images.githubusercontent.com/71774932/94028330-6efc9900-fdd9-11ea-9693-74669da9f6c2.png

mirellameelo, 2020-09-23:
@mayur444607 I always have this same issue. For somehow, and I really don't know the reason, I click in "follow camera" twice and the map just starts showing... Consider trying this simple test.

mayur444607, 2020-09-24:
@mirellameelo I click "follow camera" more then twice, but no visual data on slam is showing. any more suggestions.?

mirellameelo, 2020-09-24:
@mayur444607 I just notice by the photo you provided that the images are not arriving. So, you better debug your ROS topics.
The video/images must be arriving at the "camera/image_raw" topic. 

https://github.com/xdspacelab/openvslam/blob/9eb280aa4aad6105450debf2c85fbfd477d811f5/ros/src/openvslam/src/run_slam.cc#L57

Make sure you are publishing and republishing it as suggested in the tutorial. You can debug your topics in the terminal with  [rostopic list](http://wiki.ros.org/rostopic), or using the RVIZ software.

This is how the SLAM system looks when waiting for images:
[image]

And when the images are coming through the topic "camera/image_raw" correctly, the window "PangolinViewer: frame viewer" starts showing those, just like this:

[image]

Neel125, 2020-09-25:
@mayur444607 ,
First of run this command, 
source /path/to/openvslam/ros/devel/setup.bash
rosrun publisher video -m /path/to/video.mp4

after that open new terminal and run this command, 
rosrun image_transport republish raw in:=/video/image_raw raw out:=/camera/image_raw

and after that open new terminal and run this command,
source /path/to/openvslam/ros/devel/setup.bash
rosrun openvslam run_slam \
    -v /path/to/orb_vocab.dbow2 \
    -c /path/to/config.yaml

and also you have to run roscore in different terminal 
this helped me

mayur444607, 2020-09-25:
Visual data on slam is showing as expected, @mirellameelo @Neel125 Thank you for your suggestions, appreciate your quick responses.

ymd-stella, 2020-09-25:
@mayur444607 If your questions have been answered, please close Issue.

closed 2020-09-26 by mayur444607