xdspacelab/openvslam #456: gazebo/ros integration tutorial

wonwon0, 2020-10-09:
Hi,

I'm not sure if it's the right place to point out holes in tutorials or suggest new tutorial.

I'm trying to integrate OpenVSLAM in my gazebo simulator but i can't figure out how to do this. 

The tutorials for ROS don't seem to indicate how to make OpenVSLAM subscribe to a raw_image feed from a camera (real or simulated).

mirellameelo, 2020-10-13:
Hi.
I've been using V-REP simulator with Openvslam. Sorry but I'm not sure how to do it using Gazebo, but just in case you don't have restriction in terms of the simulator, [here](https://github.com/mirellameelo/ROS_and_V-REP) you can find a simple tutorial on how to publish images from V-REP simulator and also its respective topics, but you can change them tho to match with Openvslam topics or remmap them. 
There is examples for ROS1 and ROS2.

wonwon0, 2020-10-14:
@mirellameelo Thank you for taking the time to answer me!

I looked into your implementation and you seem to do exactly what i am trying to do!
The only piece of the puzzle that i can't seem to find is how to make OpenVSLAM take the images published in ros topics.

The examples shown in tutorials in the official doc show how to feed pre-recorded datasets but not live feeds.

I suppose i also have to generate a calibration file to give to openVSLAM about my simulated fisheye cameras. (I didn't look too much into that part yet as i'm trying to make the subcribing part work first)

mirellameelo, 2020-10-14:
@wonwon0 
What do you mean with "make the subscribing part work first"? OpenVslam already subscribes to the frames in the topic "camera/image_raw" - for a monocular camera in this case.

ROS2
https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/ros/2/src/openvslam/src/run_slam.cc#L59
ROS1
https://github.com/xdspacelab/openvslam/blob/29435e14513a437719952ba916e5994e71ea6e55/ros/1/src/openvslam/src/run_slam.cc#L57

So, once you use 
```ros2 run openvslam run_slam -v <path_to_orb_vocab> -c <path_to_YOUR_camera_config>/config.yaml ```
the SLAM system will wait for the frames, no matter where they come from. If you are interested in using a live video, for example, you can run ```ros2 run image_tools cam2image -t camera``` which gets images of a native webcam or plugged... - [Here](https://github.com/mirellameelo/openvslam/tree/ros2-base), in my personal fork (branch ros2-base), you can see the step by step for this case. 

But since you want to do that using the simulator, then you have to publish the frame on the right topic and Openvslam will be ready for you. 

I believe what you need now is: (1) form the config file of your virtual camera, and (2) from the simulator, publish the frames on the right topic (in case you decide to use V-REP for example, I think [this](https://github.com/mirellameelo/ROS_and_V-REP) can help you) OR (2) save the frames in a folder and then use ```ros2 run publisher image -i <path_to_your_frames>```. 

Let me know if I did myself clear!

wonwon0, 2020-10-16:
Woah! Awesome. I didn't know it automatically subscribed to these topics. Thank you very much for your answer.

closed 2020-10-16 by wonwon0