xdspacelab/openvslam #459: SLAM with ROS results are poor... but works fine on standard installation

PauloRodriguesJr, 2020-10-11:
I'm trying to run slam and localization on the video benchmark "aist_living_lab_1", but the performance of the algorithm drops dramatically when running from ROS nodes.
The results i'm getting when I run the code without ROS are quite good, but i am not managing to make the results from ROS do well, it keeps losing track each 2-5 seconds...
I am running on ROS melodic and using the standard OpenVSLAM installation on a clean Ubuntu 18.04.
The most wierd part of the history is that I've installed the same master branch +-6 months ago in my laptop and there slam with ROS works fine.
I'm the machine config is 6gb RAM, AMD® Phenom(tm) ii processor, and a GeForce 9600 GT GPU.
Did anybody had this issue before?
Thanks in advance.

PauloRodriguesJr, 2020-10-28:
This issue was solved. Converting it into a PR.

mirellameelo, 2020-10-28:
Hi. These days I tried to run with KITTI dataset (sequence 03) and ROS2, publishing the frames with "publisher" package from OpenVslam, but the Slam system was not able to map at all. You said you're using ROS1. Do you mind to tell a bit more what exactly was the problem, so I can try to replicate it for ROS2? Thanks in advance.

ymd-stella, 2020-10-28:
Please refer to https://spectrum.chat/openvslam/beginner/slam-with-ros-results-are-poor-but-works-fine-on-standard-installation~42725467-3590-4d69-ace7-3baa40923366.

mirellameelo, 2020-10-28:
@ymd-stella Thank you!