xdspacelab/openvslam #462: Anyone working on openvslam fusion GPS

asmaaelkeurti, 2020-10-19:
I implemented openvslam into an autopilot system on an indoor agriculture robot by using stereo camera. it works perfect. 

I am going to working on gps fusion. Before i start, i just want to know is there anyone else out there has been working on fusing the gps into openvslam. like this one https://github.com/xdspacelab/openvslam/issues/406

And I believe for the industrial usage,  fusing the gps is much more useful than fusing the imu. Cause for the low feature environment, we can just add rich feature object into the industrial environment, like QR code.  But for the global draft, i think gps would help loop detection working much better. Is it ture?

dimaxano, 2020-10-19:
Hi! This GPS fusion implementation (mentioned in #406 ) https://github.com/urbste/openvslam/tree/feat/gps-integration
works pretty well, you just need to modify some interfaces for your purposes

urbste, 2020-10-29:
But be aware that it is still in a very buggy state. Haven't found the time to work on it in a while...