xdspacelab/openvslam #465: Saved trajectory erroneous - signs flip

mama2020, 2020-10-21:
Hi!
When running run-video-slam, the saved frame_trajectory is erroneous. It seems like signs get flipped sometimes. The keyframe_trajectory looks fine.  I am using a monocular equirectangular video for the slam.  Any ideas what the reason could possibly be?
Thanks in advance. 

[image]

pgorczak, 2020-10-22:
I am having the same issue with the *socket* image pulled today but not with an older version from June. Here are evo plots of trajectories generated from the same video using the current image (773ce6881725) and the working older image (cbce321188cd).

[image]

The video came from an Insta360 Pro, recorded in 4K and then scaled to 1080p. The parameters used with both images were:

```yaml
Camera.color_order: RGB
Camera.cols: 1920
Camera.fps: 29.97002997002997
Camera.model: equirectangular
Camera.name: Insta360 Pro
Camera.rows: 1080
Camera.setup: monocular
Feature.ini_fast_threshold: 20
Feature.max_num_keypoints: 1000
Feature.min_fast_threshold: 7
Feature.num_levels: 8
Feature.scale_factor: 1.2
```

</p></details>

I can't pinpoint the commit from the working image but based on the creation date in the metadata and the build history on Docker Hub, I think it is 6de5fec058647297ca3458e4719b42947fd4a582 .

<details><summary><strong>Full metadata of working image</strong></summary><p>

```json
[
    {
        "Id": "sha256:cbce321188cd44329c3b21cc98ee325e7cfc5a6eb79cf627d448958c003bfc41",
        "RepoTags": [
            "shinsumicco/openvslam:socket"
        ],
        "RepoDigests": [
            "shinsumicco/openvslam@sha256:64fb9b6cb59b285ace12de37557e8240f5a1b2115812767ecbe1503bf209c358"
        ],
        "Parent": "",
        "Comment": "",
        "Created": "2020-06-24T02:57:41.157641548Z",
        "Container": "79a53de3dc64ff109ef5ad01d45509ac37ea140edd4c37f48e31907227442683",
        "ContainerConfig": {
            "Hostname": "79a53de3dc64",
            "Domainname": "",
            "User": "",
            "AttachStdin": false,
            "AttachStdout": false,
            "AttachStderr": false,
            "Tty": false,
            "OpenStdin": false,
            "StdinOnce": false,
            "Env": [
                "PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin",
                "DEBIAN_FRONTEND=noninteractive",
                "DEBCONF_NOWARNINGS=yes",
                "CPATH=/usr/local/include:",
                "C_INCLUDE_PATH=/usr/local/include:",
                "CPLUS_INCLUDE_PATH=/usr/local/include:",
                "LIBRARY_PATH=/usr/local/lib:",
                "LD_LIBRARY_PATH=/usr/local/lib:",
                "Eigen3_DIR=/usr/local/share/eigen3/cmake",
                "g2o_DIR=/usr/local/lib/cmake/g2o",
                "OpenCV_DIR=/usr/local/lib/cmake/opencv4",
                "DBoW2_DIR=/usr/local/lib/cmake/DBoW2",
                "sioclient_DIR=/usr/local/lib/cmake/sioclient"
            ],
            "Cmd": [
                "/bin/sh",
                "-c",
                "#(nop) ",
[... 62 lines cut ...]
            "Type": "layers",
            "Layers": [
                "sha256:b187ff70b2e47a4cf3d735dd04b4309cb3004c3069846969716b7f804a01576b",
                "sha256:5930c9e5703fbcb1f5e47dd833763a0bb0d9b1927f97b8c5e2b7e7fb4798238e",
                "sha256:c64c52ea2c16427957eb5048da0df868d445795a5dbec8fd2b594ad169295eb0",
                "sha256:ddc500d8499442f954ee7184ef015b5041629e55bf8b336304c9614154008bcd",
                "sha256:fc7805f90a9b33813c635c82fa73d1197190b4b0d46c4e6ddce03fc3f7afd280",
                "sha256:827bfe7306d748a6263352b2fd23e1dbb57d7f8eee882a8c8ac516b2be2151bf",
                "sha256:49dfa3369744d0b8cf27b616fbe74fa26ac388c227382f7d4c7eca31995876a6",
                "sha256:d675ccd9f59cf48cc230d1451bbfa7577704dee565a6aa9c1c4ab7fd0a3d19d7",
                "sha256:6dad32cc2d2e1ef75524a5c414d6f69d851ec92d269458f5979cd8ada9141def",
                "sha256:553346c861759c0851e2b293e774bd31abe0d5077e585ac0e16aea01b3c40010",
                "sha256:5279148c522584d831b7682f59f6e8f1997c8f51d491eaa9521c872c8a7261b9",
                "sha256:3efe08a6b4e0d00a1385c3dd8c3de1e33ca52e7f392c8aa53f6d06f7d6556449",
                "sha256:109ddd39b7e83a42cae67281986fa72b79fae9e685f4af810b0aeb3c2048bed9"
            ]
        },
        "Metadata": {
            "LastTagTime": "0001-01-01T00:00:00Z"
        }
    }
]
```

</p></details>

ymd-stella, 2020-11-24:
I've encountered a similar issue in https://spectrum.chat/openvslam/general/map-msg-outputs-different-positions-and-centers~34516a98-e5a3-43f7-9cf0-23b6ff19d813 and will research about it when I have time.