xdspacelab/openvslam #466: can‘t compile openvslam

ngadminq, 2020-10-22:
[image]

surfii3z, 2020-10-22:
Looks like you miss some dependencies?

Did you try [this](https://openvslam.readthedocs.io/en/master/installation.html#installing-for-linux)

`sudo apt install -y libyaml-cpp-dev libgoogle-glog-dev libgflags-dev`

ngadminq, 2020-10-22:
@surfii3z thank you,now issues are emerging
[image]

ngadminq, 2020-10-22:
@surfii3z how to deal it?
[image]

ngadminq, 2020-10-23:
root@VM-4-17-ubuntu:/home/ubuntu/openvslam# make
[  1%] Linking CXX shared library ../../lib/libopenvslam.so
/usr/bin/ld: /usr/local/lib/libyaml-cpp.a(scanscalar.cpp.o): relocation R_X86_64_PC32 against symbol `_ZNK4YAML5RegEx14MatchUncheckedINS_16StreamCharSourceEEEiRKT_' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
src/openvslam/CMakeFiles/openvslam.dir/build.make:1950: recipe for target 'lib/libopenvslam.so' failed
make[2]: *** [lib/libopenvslam.so] Error 1
CMakeFiles/Makefile2:138: recipe for target 'src/openvslam/CMakeFiles/openvslam.dir/all' failed
make[1]: *** [src/openvslam/CMakeFiles/openvslam.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
root@VM-4-17-ubuntu:/home/ubuntu/openvslam#

ymd-stella, 2020-10-30:
Please read https://openvslam.readthedocs.io/en/master/installation.html

closed 2020-11-09 by ngadminq