xdspacelab/openvslam #467: The maps created are tilted/rotated

mirellameelo, 2020-10-28:
Hi. I already tried to run OpenvSLAM using real and virtual datasets, and in both, I'm having inclined maps. As follows, a print screen from the same map, but with different angles.

I'm plotting the point clouds simply using:
```
landmark_vector = my_keyframe->get_landmarks();
...
point_pos = landmark_vector.at(i)->get_pos_in_world();
```
In red, there is the odometry info, which is set to y = 0, but its orientation also follows this weird inclination.

[image]

Does anyone is facing this issue?

Thanks in advance!

mirellameelo, 2020-10-30:
Please refer to [this link](https://spectrum.chat/openvslam/general/why-my-map-is-rotated~fe9f5065-d857-410a-be7f-de481cf1de18).

closed 2020-10-30 by mirellameelo

anthonymiglio, 2020-12-16:
Hi @mirellameelo ! I am facing the same issue and I posted my current finding at [link](https://spectrum.chat/openvslam/general/why-my-map-is-rotated~fe9f5065-d857-410a-be7f-de481cf1de18) in openvslam community.

If it interest you or you have the time, could you please check it out? Thanks