xdspacelab/openvslam #472: ROS launch file

adaCook, 2020-11-04:
I follow the tutorial of "https://openvslam.readthedocs.io/en/master/ros_package.html" and try to write a ros launch file. But here comes an error notifying that "invalied arguments".
The original operation  is to run a ros node like:
"
  rosrun openvslam run_slam \
    -v /path/to/orb_vocab.dbow2 \
    -c /path/to/config.yaml
".
And in my launch files, it is like:
"
        <node pkg="openvslam" type="run_slam" name="run_slam" >
	<param name="vocab" value=" /path/to/orb_vocab.dbow2"/>
	<param name="config" value="/path/to/config.yaml"/>
	</node>
" 
Can anybody  point out what problem is?

adaCook, 2020-11-04:
Sorry for the fomat error. codes in my launch file is like:
"        <node pkg="openvslam" type="run_slam" name="run_slam" >
	<param name="vocab" value=" /path/to/orb_vocab.dbow2 "/>
	<param name="config" value="/path/to/config.yaml"/>
	</node>
"

adaCook, 2020-11-04:
Sorry for the format error. codes of ros launch file is like:
[image]

dimaxano, 2020-11-13:
Hi! To pass arguments via roslaunch they should be a ros params. But in openvslam case, they are ussual command line parameters
You can read more about ros params here: http://wiki.ros.org/roscpp_tutorials/Tutorials/Parameters
Also, field `type` should be equal to `node` in your case