xdspacelab/openvslam #480: What causes the points to be dropped?

JockTokk, 2020-11-24:
Hi!
Have you faced [this](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/188#issue-746389841) problem?
Thanks for reply!

dimaxano, 2020-11-25:
Hi!
If I understood you correctly, this passage from ORB-SLAM paper should explain landmarks disappearing

[image]

JockTokk, 2020-11-25:
@dimaxano thanks. I'm not sure if this is the solution because only the points that are behind the camera disappear.

ymd-stella, 2020-11-25:
The issue is not for OpenVSLAM.
I think it's good to ask at https://spectrum.chat/openvslam/general, but it's inappropriate to make an Issue here.

JockTokk, 2020-11-26:
@ymd-stella ok, thanks.

closed 2020-11-26 by JockTokk