xdspacelab/openvslam #481: How to only save the new keypoints after run localization with mapping flag on?

asmaaelkeurti, 2020-11-26:
In one of my project, i saved map. Run localization the mapping couple times. I noticed the map msg file gets bigger. 

I was wondering what would be the best way to save the new keypoints and keyframe and drop the old ones. 

i am pretty familiar with the source code. Just need some one point me the right direction.

JoaoAVDuarte, 2020-11-26:
>  I noticed the map msg file gets bigger.

How much bigger does it get? If it's a few kilobytes or megabites it's the usual behaviour.
Localization with the mapping flag turned on should pick up roughly the same keypoints as `run_video_slam` without much difference, and it re-writes the keypoints over the previous ones. It should not duplicate the data.
Also, consider using Spectrum's [OpenVSLAM community](https://spectrum.chat/openvslam).

asmaaelkeurti, 2020-11-28:
The real world environment keeps changing everyday. I ran run_camera_slam. The live stereo camera.  Not the same video over and over again. What i noticed is that mapping module keeps getting the new landmark, but didn't remove the old landmark not observable. I think that's why the map file gets bigger. Is there a way to remove the old landmark that not observed?