xdspacelab/openvslam #482: *** Error in `./run_image_slam': core dump

fangice, 2020-11-28:
When I execute with my own image sequence the error below appears.

Command line:
./run_image_slam -v ../config/config_files/orb_vocab.dbow2 -c ../config/config_files/indoor_config.yaml -i /home/fhy/flying/data/image1127

Output:

[2020-11-28 15:18:10.047] [I] config file loaded: ../config/config_files/indoor_config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: indoor monocular
- setup: Monocular
- fps: 20
- cols: 1280
- rows: 720
- color: RGB
- model: Perspective
  - fx: 945.982
  - fy: 947.034
  - cx: 639.95
  - cy: 361.204
  - k1: 0
  - k2: 0
  - p1: -0
  - p2: 0
  - k3: 0
  - min x: 0
  - max x: 1280
  - min y: 0
  - max y: 720
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7

[2020-11-28 15:18:10.049] [I] loading ORB vocabulary: ../config/config_files/orb_vocab.dbow2
[2020-11-28 15:18:10.380] [I] startup SLAM system
[2020-11-28 15:18:10.380] [I] start mapping module
[2020-11-28 15:18:10.380] [I] start global optimization module
[2020-11-28 15:18:18.467] [I] initialization succeeded with H
[2020-11-28 15:18:18.507] [I] new map created with 467 points: frame 0 - frame 87
*** Aborted at 1606547918 (unix time) try "date -d @1606547918" if you are using GNU date ***
PC: @     0x7fa44abe2472 openvslam::tracking_module::update_local_keyframes()
*** SIGSEGV (@0x450) received by PID 14671 (TID 0x7fa4292d9700) from PID 1104; stack trace: ***
    @     0x7fa4480974c0 (unknown)
    @     0x7fa44abe2472 openvslam::tracking_module::update_local_keyframes()
    @     0x7fa44abe2d49 openvslam::tracking_module::update_local_map()
    @     0x7fa44abe4045 openvslam::tracking_module::track()
    @     0x7fa44abe43bf openvslam::tracking_module::track_monocular_image()
    @     0x7fa44abcd97c openvslam::system::feed_monocular_frame()
    @           0x417fec _ZNSt6thread5_ImplISt12_Bind_simpleIFZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESF_SF_jbbbSF_EUlvE_vEEE6_M_runEv
    @     0x7fa4486fac80 (unknown)
    @     0x7fa44a9126ba start_thread
    @     0x7fa4481694dd clone
段错误 (核心已转储)

Has anybody ever met?

ymd-stella, 2020-11-28:
The code you are using seems to be outdated. Please check if it reproduces in the latest master branch.

Does the problem occur every time or does it occur occasionally?

It is recommended to use cmake with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` to identify the location of the error.

fangice, 2020-11-28:
> The code you are using seems to be outdated. Please check if it reproduces in the latest master branch.
> 
> Does the problem occur every time or does it occur occasionally?
> 
> It is recommended to use cmake with `-DCMAKE_BUILD_TYPE=RelWithDebInfo` to identify the location of the error.

Thank you very much for your suggestion. By compiling the latest version of the project, the problem has been solved!

ymd-stella, 2020-11-28:
@fangice Please close this issue.

closed 2020-11-28 by fangice