xdspacelab/openvslam #487: Regarding 4k image sequence visual SLAM

RunqiuBao, 2020-11-30:
Hi, thanks for this great library.

I wanted to run a 4k image sequence with openvslam, but I found the result trajectory accuracy not so good. (worse than using 1k image)
I suppose there must be some parameter tuning before changing to 4k image.

Then I adjusted the orb parameters in  the config file. 
I also adjusted a searching range parameter of orb feature matching hard-coded into stereo.cc.
The result trajectory accuracy improved, but still it can not  surpass when using 1k images.
Theoretically, 4k image result should be better than 1k image result, right?
I wonder if there are more hard-coded parameters that I can optimize for running on a 4k image sequence.

Best regards.
Runqiu

ryleymcc, 2020-12-11:
I'm using a 4k ricoh theta over usb for real-time slam and it is very accurate

RunqiuBao, 2020-12-12:
> I'm using a 4k ricoh theta over usb for real-time slam and it is very accurate

How accurate is it? Did you have a quantitative evaluation?
And have you modified the SLAM parameters as well as orb parameters?