xdspacelab/openvslam #488: How to save depth data and how to write camera config when using own RGBD datasets (taken by Realsense D435i)

Shunichi09, 2020-11-30:
Hi!
Thank you for your great work and code.

I'm trying to use my RGBD datasets.
I followed TUM datasets format. 
I could run the ./run_tum_rgbd_slam and got the trajectory, but the trajectory was quite small. (e.g. having 0.001).

I think the problem was caused by how I save the depth image as .png and how I set camera config (yaml file).

so, the questions are

1. How should we save depth image as png file ??

Currently, I'm using opencv and save it as follows:

```py
for topic, msg, t in rosbag.Bag(bag_file_path).read_messages():
        time_stamp = t.to_nsec()
        time_stamp = str(time_stamp)[:-9] + "." + str(time_stamp)[-9:-3]
        bridge = CvBridge()

        if topic == "/camera/color/image_raw":
            bgr8_img = bridge.imgmsg_to_cv2(msg, "bgr8")

            image_file_name = str(time_stamp) + ".png"
            cv2.imwrite(os.path.join(save_dir, "rgb", image_file_name), bgr8_img)
            write_txtfile(os.path.join(save_dir, "rgb.txt"), str(time_stamp) + " " + "rgb/" + image_file_name + "\n")

        if topic == "/camera/aligned_depth_to_color/image_raw":
            depth_img = bridge.imgmsg_to_cv2(msg, "passthrough")
            # I should do something to depth image ??
            depth_img = np.array(depth_img, dtype=np.int)

            image_file_name = str(time_stamp) + ".png"
            cv2.imwrite(os.path.join(save_dir, "depth", image_file_name), depth_img)
            write_txtfile(os.path.join(save_dir, "depth.txt"), str(
                time_stamp) + " " + "depth/" + image_file_name + "\n")
```

2.  What is camera.focal_x_baseline and How can we get this parameter in D435i (Realsense) ??

My D435i's yaml file is as follows;:

```
#==============#
# Camera Model #
#==============#

Camera.name: "Realsense D435i"
Camera.setup: "RGBD"
Camera.model: "perspective"

Camera.fx: 609.4935302734375
Camera.fy: 609.4905395507812
Camera.cx: 323.1676940917969
Camera.cy: 240.14817810058594

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

Camera.fps: 30.0
Camera.cols: 640
Camera.rows: 480
Camera.focal_x_baseline: 40.0  # What is this??

Camera.color_order: "RGB"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 1000
Feature.scale_factor: 1.2
Feature.num_levels: 8
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 10.0
depthmap_factor: 1000.0

#===========================#
# PangolinViewer Parameters #
#===========================#

PangolinViewer.keyframe_size: 0.05
PangolinViewer.keyframe_line_width: 1
PangolinViewer.graph_line_width: 1
PangolinViewer.point_size: 2
PangolinViewer.camera_size: 0.08
PangolinViewer.camera_line_width: 3
PangolinViewer.viewpoint_x: 0
PangolinViewer.viewpoint_y: -0.9
PangolinViewer.viewpoint_z: -1.9
PangolinViewer.viewpoint_f: 400
```

I really appreciate your help.

ymd-stella, 2020-11-30:
You should ask your question at https://spectrum.chat/openvslam/beginner.

Shunichi09, 2020-11-30:
Thank you for your advise. I posted my issue to the chat.

closed 2020-12-03 by Shunichi09