xdspacelab/openvslam #492: ROS 2 Colcon build fail

mathislm, 2020-12-02:
I'm following [this guide](https://openvslam.readthedocs.io/en/develop/ros2_package.html#installation) for the ROS2 package and on this step I get this error : 

```
mathis@mathis-XPS:~/cxxLibs/openvslam/ros$ sudo colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_transport
Starting >>> camera_calibration_parsers
Starting >>> image_geometry
--- stderr: cv_bridge
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_ros", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_ros"
  with any of the following names:
    ament_cmake_rosConfig.cmake
    ament_cmake_ros-config.cmake

  Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
  set "ament_cmake_ros_DIR" to a directory containing one of the above files.
  If "ament_cmake_ros" provides a separate development package or SDK, be
  sure it has been installed.

Failed   <<< cv_bridge [0.08s, exited with code 1]
--- stderr: image_transport
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findament_cmake_ros.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ament_cmake_ros", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake_ros"
  with any of the following names:

    ament_cmake_rosConfig.cmake
    ament_cmake_ros-config.cmake

  Add the installation prefix of "ament_cmake_ros" to CMAKE_PREFIX_PATH or
  set "ament_cmake_ros_DIR" to a directory containing one of the above files.
  If "ament_cmake_ros" provides a separate development package or SDK, be
  sure it has been installed.

Failed   <<< image_transport [0.07s, exited with code 1]
Aborted  <<< camera_calibration_parsers [0.07s]
Aborted  <<< image_geometry [0.06s]

Summary: 0 packages finished [0.20s]
  2 packages failed: cv_bridge image_transport
  2 packages aborted: camera_calibration_parsers image_geometry
  4 packages had stderr output: camera_calibration_parsers cv_bridge image_geometry image_transport
  6 packages not processed
```
I read issue #432 but I didn't understood everything, and it's not really the same problem I think. Any idea why this is failling ?
Thank you

mirellameelo, 2020-12-02:
Hi. Did you source your environment before using 'colcon build'?
      
       source /opt/ros/dashind/setup.bash

mathislm, 2020-12-02:
Yes I did, though I'm using ros foxy, maybe it's the reason why this is not working.
In any case, I followed [this tutorial](https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Development-Setup/) to install ros2 foxy, and so I sourced the environment with 
```source ~/ros2_foxy/install/setup.bash```
I also tried installing ros2 with apt and then source with
```source /opt/ros/foxy/setup.bash```
that didn't fixed it

I noticed however that both setup.bash are quite different

```
cat /opt/ros/foxy/setup.bash 
# copied from ament_package/template/prefix_level/setup.bash

AMENT_SHELL=bash

# source setup.sh from same directory as this file
AMENT_CURRENT_PREFIX=$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" && pwd)
# trace output
if [ -n "$AMENT_TRACE_SETUP_FILES" ]; then
  echo ". \"$AMENT_CURRENT_PREFIX/setup.sh\""
fi
. "$AMENT_CURRENT_PREFIX/setup.sh"
```

```
cat ~/ros2_foxy/install/setup.bash 
# generated from colcon_bash/shell/template/prefix_chain.bash.em

# This script extends the environment with the environment of other prefix
# paths which were sourced when this file was generated as well as all packages
# contained in this prefix path.

# function to source another script with conditional trace output
# first argument: the path of the script
_colcon_prefix_chain_bash_source_script() {
  if [ -f "$1" ]; then
    if [ -n "$COLCON_TRACE" ]; then
      echo ". \"$1\""
    fi
    . "$1"
  else
    echo "not found: \"$1\"" 1>&2
  fi
}

# source chained prefixes
# setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
COLCON_CURRENT_PREFIX="/opt/ros/foxy"
_colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"
# setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
COLCON_CURRENT_PREFIX="/home/mathis/ros2_ws/install"
_colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"

# source this prefix
# setting COLCON_CURRENT_PREFIX avoids determining the prefix in the sourced script
COLCON_CURRENT_PREFIX="$(builtin cd "`dirname "${BASH_SOURCE[0]}"`" > /dev/null && pwd)"
_colcon_prefix_chain_bash_source_script "$COLCON_CURRENT_PREFIX/local_setup.bash"

unset COLCON_CURRENT_PREFIX
unset _colcon_prefix_chain_bash_source_script
```

is this normal ?

I tried using ros2 with the simple example of the talker and listener and it's working properly. I also managed to install the ROS2 package for the Intel RealSense D435i and it worked fine, so I suppose the source is working fine... maybe ?

Thanks or the help

mirellameelo, 2020-12-02:
I also think the source step is fine. Just to be sure, you said you followed the "official" tutorial, but it says:

       git clone -b **dashing** --single-branch https://github.com/ros-perception/image_common.git

Did you try to build with the ros2 branch instead?

mathislm, 2020-12-02:
I tried using the ros2 branch, it gives the same errors.
What I don't understand is that, even if I'm not using the proper version, which is probably like you said an issue I'll have to tackle later, it just doesn't find packages. How come ? I don't exactly know where to look for them myself, but I'm quite sure they are indeed installed.
Maybe I could check that first, do you have some clues on how to do that ?

ymd-stella, 2020-12-02:
It's weird that sudo is used to do `colcon build`.
Clearly you must be doing something wrong about ros2.

This is not a issue with OpenVSLAM, it's your personal issue.
It is acceptable to ask questions at https://spectrum.chat/openvslam/beginner, but it is annoying to ask them here.

mirellameelo, 2020-12-02:
Yeah... The error you posted is the same one I get when I forget to source the environment, that's why I thought it could be something related to it. Maybe you could try to build from scratch a package, just to make sure. Take a look [here](https://index.ros.org/doc/ros2/Tutorials/Creating-Your-First-ROS2-Package/). 
There is some guidance if you look for something like "workspace properly overlaid ros2". In the last case, I'd try to reinstall (Debian pkg - safer) the foxy version.

closed 2021-01-04 by mathislm