xdspacelab/openvslam #500: Pangolin veiwer follow camera

ryleymcc, 2020-12-11:
When running camera slam with pangolin viewer, the map doesn't start until I toggle "follow camera" a few times and then it won't follow the camera properly. If i move the camera forward, pangolin moves backward, if i go left it goes right, but up and down works and so does rotation.

dimaxano, 2020-12-11:
Hi! I also met this problem
As I understand Pangolin Viewer displays [camera to world pose](https://github.com/xdspacelab/openvslam/blob/9eb280aa4aad6105450debf2c85fbfd477d811f5/src/pangolin_viewer/viewer.cc#L185), that is why all movements are inverse

Can you please specify in what demo you see no inverse movement on tha map according to real movement?

ryleymcc, 2020-12-11:
Sorry about that i guess i was thinking about a video i was watching. The demos DO NOT follow the camera pose

dimaxano, 2020-12-12:
I do not have a working solution for now, unfortunately. I am just using workaround you described above