xdspacelab/openvslam #504: CameraToWorld pose does not similar to WorldToCamera pose

dimaxano, 2020-12-17:
Hi!
I found that pose world->camera doesn't similar to camera->world pose ([plot](https://imgur.com/a/qNnGjkL), blue line - camera->world, pink - world->camera), but as I understand they should differ only by SE(3) transformation. To my view, camera->world pose is more correct and represents real vehicle movement better.

Also I see that camera->world pose used everywhere in openvslam where we need trajectory representation like in pangolin_viewer and trajectory_io module.

Can someone explain me why those positions differs so much? @shinsumicco ?

dimaxano, 2020-12-21:
Hi, @ymd-stella 
Sorry for bothering
I see you answers a lot here, maybe you can explain what is going on in my case. I am really confused with that

ymd-stella, 2020-12-21:
I don't have time for this.
Please learn about 3D Pose representation by yourself.

jcmonteiro, 2020-12-21:
@dimaxano they differ only by a homogenous transformation, so there is no deformation. You did not provide any information other than those two plots, so I'm struggling to understand even what it is that you want to solve.

Try to provide more context on your setup, how to reproduce and a better explanation of the issue. Otherwise, contributors will not have the time to assist you.