xdspacelab/openvslam #507: No map is being created

jediofgever, 2020-12-21:
Hi ,
 I am trying to run this in simulation with ros2. The image has quite some points detected but there is nothing in the 3D map visualizer.  
I use the command; 
`ros2 run openvslam run_slam     -v /home/ros2-foxy/orb_vocab.dbow2 -c /home/ros2-foxy/config.yaml`

[image]

The config file is; 

```
#==============#
# Camera Model #
#==============#

Camera.name: "Realsense D435i"
Camera.setup: "monocular"
Camera.model: "perspective"

Camera.fx: 640.83978271484375
Camera.fy: 640.865234375
Camera.cx: 640.91400146484375
Camera.cy: 360.83836364746094

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

Camera.fps: 30.0
Camera.cols: 1280
Camera.rows: 720
Camera.focal_x_baseline: 22.942 # 5 cm

Camera.color_order: "BGR"

#================#
# ORB Parameters #
#================#

Feature.max_num_keypoints: 2600
Feature.scale_factor: 1.2
Feature.num_levels: 11
Feature.ini_fast_threshold: 20
Feature.min_fast_threshold: 7

#=====================#
# Tracking Parameters #
#=====================#

depth_threshold: 200.0 # depth_threshold * base_line[m] = 10m
depthmap_factor: 1000.0
```

All of this is run under simulatio. Do you have any idea what th map wouldn't be created ?

surfii3z, 2020-12-21:
Your openvslam is not initialize yet. I think it might be that your resolution in the config file doesn't match with that passed to openVSLAM

jediofgever, 2020-12-21:
fter I movecamere I it is initilized but the program dies right after 
```

[2020-12-21 22:08:25.206] [I] loading ORB vocabulary: /home/ros2-foxy/orb_vocab.dbow2
[2020-12-21 22:08:25.574] [I] startup SLAM system
[2020-12-21 22:08:25.574] [I] start mapping module
[2020-12-21 22:08:25.574] [I] start global optimization module
[2020-12-21 22:08:30.569] [I] initialization succeeded with F
[2020-12-21 22:08:30.613] [I] new map created with 120 points: frame 0 - frame 59
```

surfii3z, 2020-12-21:
Please try to run with --debug flag. Again check the resolution of your RGB image.

jediofgever, 2020-12-21:
no really useful with information with  `--debug ` unfortunately 
```

[2020-12-21 22:20:36.028] [I] loading ORB vocabulary: /home/ros2-foxy/orb_vocab.dbow2
[2020-12-21 22:20:36.434] [D] CONSTRUCT: data::camera_database
[2020-12-21 22:20:36.434] [D] CONSTRUCT: data::map_database
[2020-12-21 22:20:36.434] [D] CONSTRUCT: data::bow_database
[2020-12-21 22:20:36.434] [D] CONSTRUCT: publish::frame_publisher
[2020-12-21 22:20:36.434] [D] CONSTRUCT: publish::map_publisher
[2020-12-21 22:20:36.434] [D] CONSTRUCT: module::initializer
[2020-12-21 22:20:36.434] [D] CONSTRUCT: module::relocalizer
[2020-12-21 22:20:36.434] [D] CONSTRUCT: tracking_module
[2020-12-21 22:20:36.434] [D] CONSTRUCT: mapping_module
[2020-12-21 22:20:36.434] [D] CONSTRUCT: global_optimization_module
[2020-12-21 22:20:36.434] [I] startup SLAM system
[2020-12-21 22:20:36.434] [I] start mapping module
[2020-12-21 22:20:36.434] [I] start global optimization module
[2020-12-21 22:20:38.795] [D] CONSTRUCT: initialize::perspective
[2020-12-21 22:21:03.196] [I] initialization succeeded with F
[2020-12-21 22:21:03.196] [D] DESTRUCT: initialize::perspective
[2020-12-21 22:21:03.245] [I] new map created with 125 points: frame 0 - frame 289
```

surfii3z, 2020-12-21:
Then it's calibration problem I guess.

jediofgever, 2020-12-21:
I don't think it is because of that. I am using simulated camera that comes with word of god calibration info

mirellameelo, 2020-12-21:
Please, refer to: https://openvslam.readthedocs.io/en/master/simple_tutorial.html

[image]

jediofgever, 2020-12-21:
I need develop branch since I am on ROS2 , does this also applies to develop branch ?

closed 2020-12-22 by jediofgever