xdspacelab/openvslam #515: Openvslam + ROS + stereo

kevkid, 2021-01-15:
Is there a way for me to use openvslam for a sets of image pairs as if it were a stereo camera? For example if I have a left image and a right image, can I use these two images as a stereo image and run openvslam on it? Would I be able to do this with the ros wrapper?

dimaxano, 2021-01-18:
Generally, yes, you can do SLAM on stereo images with openvslam (images should be prerectified and undistorted or you need intrinsic and extrinsic camera system parameters to do that). But it is currently not implemented in ROS wrapper, but you can do it yourself following [this](https://github.com/xdspacelab/openvslam/blob/9eb280aa4aad6105450debf2c85fbfd477d811f5/example/run_kitti_slam.cc#L143) example

mirellameelo, 2021-01-18:
Hi. The SLAM system uses a stereo pair of images to estimate depth, and then, map. To estimate it, a triangulation step is necessary, which requires the images to be rectified. So, are your cameras horizontally aligned with each other? Are the baseline (the distance between the lens) constant? Is the resolution the same for both images? Do they have the same opening angle? If it not, then you have to "prepare" your images before use them as input to Openvslam system.

kevkid, 2021-01-18:
> Hi. The SLAM system uses a stereo pair of images to estimate depth, and then, map. To estimate it, a triangulation step is necessary, which requires the images to be rectified. So, are your cameras [...]

All excellent questions. I will be taking the necessary steps to preprocess them. But it looks like the ros wrapper does not have stereo implemented :(

kevkid, 2021-01-18:
> Generally, yes, you can do SLAM on stereo images with openvslam (images should be prerectified and undistorted or you need intrinsic and extrinsic camera system parameters to do that). But it is [...]

Thank you, I will give this a try!

mirellameelo, 2021-01-18:
@kevkid 
Basically what I had to do in order to use my stereo camera was to add a new condition for the stereo setup, such as:
https://github.com/mirellameelo/rtabmap_openvslam/blob/7be49f95229691bc4e0d44057dd2da8d5276f231/ros/2/src/openvslam/src/run_slam.cc#L382

Subscribe not one, but the two images (I'm considering they are synchronized), such as:
https://github.com/mirellameelo/rtabmap_openvslam/blob/7be49f95229691bc4e0d44057dd2da8d5276f231/ros/2/src/openvslam/src/run_slam.cc#L256

And call "feed_stereo_frame" instead of "feed_monocular_frame", such as:
https://github.com/mirellameelo/rtabmap_openvslam/blob/7be49f95229691bc4e0d44057dd2da8d5276f231/ros/2/src/openvslam/src/run_slam.cc#L158