xdspacelab/openvslam #523: *** Aborted at 1613232792 (unix time) try "date -d @1613232792" if you are using GNU date ***

Jerichoxiii, 2021-02-13:
~/src/openvslam/openvslam-pangolinViewer/build$ ./run_video_slam -v ./orb_vocab/orb_vocab.dbow2 -m ./aist_living_lab_1/video.mp4 -c ./aist_living_lab_1/config.yaml --frame-skip 3 --no-sleep --map-db map.msg  date -d @1613232770
[2021-02-13 18:13:11.805] [I] config file loaded: ./aist_living_lab_1/config.yaml
  ___               __   _____ _      _   __  __ 
 / _ \ _ __  ___ _ _\ \ / / __| |    /_\ |  \/  |
| (_) | '_ \/ -_) ' \\ V /\__ \ |__ / _ \| |\/| |
 \___/| .__/\___|_||_|\_/ |___/____/_/ \_\_|  |_|
      |_|                                        

Copyright (C) 2019,
National Institute of Advanced Industrial Science and Technology (AIST)
All rights reserved.

This is free software,
and you are welcome to redistribute it under certain conditions.
See the LICENSE file.

Camera Configuration:
- name: RICOH THETA S 960
- setup: Monocular
- fps: 30
- cols: 1920
- rows: 960
- color: RGB
- model: Equirectangular
ORB Configuration:
- number of keypoints: 2000
- scale factor: 1.2
- number of levels: 8
- initial fast threshold: 20
- minimum fast threshold: 7
- mask rectangles:
  - [0, 1, 0, 0.1]
  - [0, 1, 0.84, 1]
  - [0, 0.2, 0.7, 1]
  - [0.8, 1, 0.7, 1]

[2021-02-13 18:13:11.805] [I] loading ORB vocabulary: ./orb_vocab/orb_vocab.dbow2
[2021-02-13 18:13:12.121] [I] startup SLAM system
[2021-02-13 18:13:12.121] [I] start mapping module
[2021-02-13 18:13:12.121] [I] start global optimization module
[2021-02-13 18:13:12.344] [I] initialization succeeded with E
double free or corruption (out)
*** Aborted at 1613232792 (unix time) try "date -d @1613232792" if you are using GNU date ***
PC: @     0x7f233a03718b gsignal
*** SIGABRT (@0x3e800002889) received by PID 10377 (TID 0x7f23146fb700) from PID 10377; stack trace: ***
    @     0x7f233c762631 (unknown)
    @     0x7f233c4db3c0 (unknown)
    @     0x7f233a03718b gsignal
    @     0x7f233a016859 abort
    @     0x7f233a0813ee (unknown)
    @     0x7f233a08947c (unknown)
    @     0x7f233a08b120 (unknown)
    @     0x7f2337ef5bf2 g2o::OptimizableGraph::~OptimizableGraph()
    @     0x7f233c676d2b openvslam::optimize::global_bundle_adjuster::optimize()
    @     0x7f233c627618 openvslam::module::initializer::create_map_for_monocular()
    @     0x7f233c628163 openvslam::module::initializer::initialize()
    @     0x7f233c5a5dd0 openvslam::tracking_module::initialize()
    @     0x7f233c5a781d openvslam::tracking_module::track()
    @     0x7f233c5a7c26 openvslam::tracking_module::track_monocular_image()
    @     0x7f233c59c020 openvslam::system::feed_monocular_frame()
    @     0x55c3e04c01dc _ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJZ13mono_trackingRKSt10shared_ptrIN9openvslam6configEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESG_SG_jbbbSG_EUlvE_EEEEE6_M_runEv
    @     0x7f233a2d6d84 (unknown)
    @     0x7f233c4cf609 start_thread
    @     0x7f233a113293 clone
Aborted (core dumped)

Jerichoxiii, 2021-02-13:
Hello there ,
I am trying to run the OpenvSlam with PangolinoViewer but i found the error above while I am trying to run the application .
Thanks in advance

dimaxano, 2021-02-15:
Hi! I would advise building openvslam with `-DCMAKE_BUILD_TYPE=RelWithDebInfo`
Maybe that would print some helpful messages

Jerichoxiii, 2021-02-15:
~/src/openvslam/openvslam-pangolinViewer/build$ cmake ..
-- Build type: RelWithDebInfo
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): ENABLED
-- Use OpenCV 4.5.1
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /opt/ros/noetic/lib/cmake/g2o/g2oConfig.cmake:4 (find_dependency)
  src/openvslam/CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- OpenMP: ENABLED
-- SSE3 for ORB extraction (-msse3): ENABLED
-- SSE for floating-point operation (-mfpmath=sse): ENABLED
-- BoW framework: DBoW2 (found in /usr/local/include)
-- Found protoc executable: /usr/bin/protoc
-- No preference for use of exported glog CMake configuration set, and no hints for include/library directories provided. Defaulting to preferring an installed/exported glog CMake configuration if available.
-- Failed to find installed glog CMake configuration, searching for glog build directories exported with CMake.
-- Failed to find an installed/exported CMake configuration for glog, will perform search for installed glog components.
-- Stack trace logger: ENABLED
-- Google Perftools: DISABLED
-- Viewer for examples: PangolinViewer
-- Downloading/updating googletest
-- Configuring done
-- Generating done
-- Build files have been written to: /home/jericho/src/openvslam/openvslam-pangolinViewer/build/googletest-download
Scanning dependencies of target googletest-download
[100%] Built target googletest-download
-- Configuring done
-- Generating done

Jerichoxiii, 2021-02-15:
Unfortunately the same story. It did compile successfully but it has exactly the same run time error. I suspect that there is something wrong with the dependencies. For example, Do  I have to build the g2o and DBoW2 with any additional settings?

dimaxano, 2021-02-15:
Did you follow the instruction for building g2o and DBoW2 from openvslam docs?

Jerichoxiii, 2021-02-15:
cmake ..
-- Build type: Release
-- ccache: Disabled
-- Address/Memory sanitizer: DISABLED
-- Architecture-aware optimization (-march=native): DISABLED
-- Use OpenCV 4.5.1
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /opt/ros/noetic/lib/cmake/g2o/g2oConfig.cmake:4 (find_dependency)
  src/openvslam/CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Failed to find CXSparse - Could not find CXSparse include directory, set CXSPARSE_INCLUDE_DIR to directory containing cs.h
-- Found AMD library: /usr/lib/x86_64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/x86_64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/x86_64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/x86_64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found SuiteSparse_config library: /usr/lib/x86_64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/x86_64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- OpenMP: DISABLED
-- SSE3 for ORB extraction (-msse3): DISABLED
-- SSE for floating-point operation (-mfpmath=sse): DISABLED
-- BoW framework: DBoW2 (found in /usr/local/include)
-- Stack trace logger: DISABLED
-- Google Perftools: DISABLED
-- Viewer for examples: PangolinViewer
-- Configuring done
-- Generating done
-- Build files have been written to:

Jerichoxiii, 2021-02-15:
It does work now with this build now. Yes, I followed the documentation, I am just asking you because the instructions are for the 16.04 ubuntu and now I have 20.04. 
Now It creates the map and localizes the camera position as It is in the documentation but there is an issue with the camera view. When I press to follow the camera on the settings of the Map view window, there is a huge bias on the view. It moves but it is out of the scope in both cases. When it does the localization I can not understand where the camera is. I can see only the color change of the landmarks.
Thanks

Jerichoxiii, 2021-02-15:
https://user-images.githubusercontent.com/27870611/107959904-a51aff00-6fac-11eb-8a9d-e0d5b1d0e223.mp4

Jerichoxiii, 2021-02-15:
https://user-images.githubusercontent.com/27870611/107960189-fc20d400-6fac-11eb-93a7-b0c7368b8b11.mp4

https://user-images.githubusercontent.com/27870611/107960207-02af4b80-6fad-11eb-8903-cca5fb426f13.mp4

I hope these videos help you understand what I mean.
Thanks

dimaxano, 2021-02-16:
Hi! I'm glad that the initial error has gone.

I also experience the same behavior of PangolinViewer. But I did nothing with that :)

dimaxano, 2021-02-16:
Just thought about a kind of workaround in your case: you may change the color of the camera in Pangolin (change values of the color [here](https://github.com/xdspacelab/openvslam/blob/9eb280aa4aad6105450debf2c85fbfd477d811f5/src/pangolin_viewer/color_scheme.cc#L33) to `1.0f, 1.0f, 0.0f`, it will be yellow now) and increase frame size in the Map Viewer window
After that, it is easier to find the camera while localization